BitMaker-hub/NerdMiner_v2 #1: Possible to run on other ESP32 versions

PsySc0rpi0n, 2023-03-28:
Hi.
Is this possible to be run in other versions of ESP32 such as ESP32 DevKitC-32E, which is based in the following iiem:
https://www.espressif.com/en/products/devkits/esp32-devkitc
and based on this part:
https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32e_esp32-wroom-32ue_datasheet_en.pdf

PsySc0rpi0n, 2023-03-28:
I tried to compile the repository to a normal / single ESP32-DevKitC-32E, adding the required libraries, one by one, by hand, and compiling with `arduino-cli` for the FQBN `esp32:esp32:esp32` but it ended up with too big binaries.

Here is the final part of the compilation proccess:
```
python /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/gen_esp32part.py -q /tmp/arduino/sketches/E17AC8752314FCAA762532757744502B/partitions.csv /tmp/arduino/sketches/E17AC8752314FCAA762532757744502B/NerdMinerV2.ino.partitions.bin
Multiple libraries were found for "WiFi.h"
  Used: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
  Not used: /home/psysc0rpi0n/Arduino/libraries/WiFiNINA
Using library WiFi at version 1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi 
Using library WebServer at version 1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WebServer 
Using library TFT_eSPI at version 2.5.23 in folder: /home/psysc0rpi0n/Arduino/libraries/TFT_eSPI 
Using library SPI at version 1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPI 
Using library FS at version 1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/FS 
Using library SPIFFS at version 1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPIFFS 
Using library OpenFontRender at version 1.0.0 in folder: /home/psysc0rpi0n/Arduino/libraries/OpenFontRender 
Using library ArduinoJson at version 6.21.1 in folder: /home/psysc0rpi0n/Arduino/libraries/ArduinoJson 
Using library WiFiManager at version 2.0.15-rc.1 in folder: /home/psysc0rpi0n/Arduino/libraries/WiFiManager 
Using library Update at version 1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Update 
Using library DNSServer at version 1.1.0 in folder: /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/DNSServer 
/home/psysc0rpi0n/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-size -A /tmp/arduino/sketches/E17AC8752314FCAA762532757744502B/NerdMinerV2.ino.elf
Sketch uses 1357754 bytes (103%) of program storage space. Maximum is 1310720 bytes.
Global variables use 40848 bytes (12%) of dynamic memory, leaving 286832 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

Used library   Version     Path                                                                                
WiFi           1.0         /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi     
WebServer      1.0         /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WebServer
TFT_eSPI       2.5.23      /home/psysc0rpi0n/Arduino/libraries/TFT_eSPI                                        
SPI            1.0         /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPI      
FS             1.0         /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/FS       
SPIFFS         1.0         /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/SPIFFS   
OpenFontRender 1.0.0       /home/psysc0rpi0n/Arduino/libraries/OpenFontRender                                  
ArduinoJson    6.21.1      /home/psysc0rpi0n/Arduino/libraries/ArduinoJson                                     
WiFiManager    2.0.15-rc.1 /home/psysc0rpi0n/Arduino/libraries/WiFiManager                                     
Update         1.0         /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Update   
DNSServer      1.1.0       /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/DNSServer

Used platform Version Path                                                            
esp32:esp32   1.0.6   /home/psysc0rpi0n/.arduino15/packages/esp32/hardware/esp32/1.0.6
Error during build: text section exceeds available space in board
psysc0rpi0n@psyHost:~/Downloads/arduino-cli-latest$ ls -lah /tmp/arduino/sketches/E17AC8752314FCAA762532757744502B/NerdMinerV2.ino.elf
-rwxr-xr-x 1 psysc0rpi0n psysc0rpi0n 17M Mar 28 23:11 /tmp/arduino/sketches/E17AC8752314FCAA762532757744502B/NerdMinerV2.ino.elf
```

BitMaker-hub, 2023-03-28:
Hi,

You can choose one of these three things to solve this problem:
1. Change the board partitions to have more memory for your program. On this project I use the partition scheme "huge APP" 3M flash 1 M SPIFF. You seem being using 1.5MB for APP (not enought)
2. Delete all TFT lines not used in your case. Will save you space of images.
3. You can use first version of NerdMiner prepared to work directly on a board like you pretend with minimal usage of TFT and with wifi and miner settings preflashed (config.h)

closed 2023-03-28 by BitMaker-hub

BitMaker-hub, 2023-03-29:
Problem still seems the same, as appears:
`Sketch uses 1357754 bytes (**103%**) of program storage space. Maximum is 1310720 bytes.`

**Sketch too big**; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

I would suggest an easy solution, comment the following line   `tft.pushImage(0, 0, initWidth, initHeight, initScreen);
`This should let you build and load it

PsySc0rpi0n, 2023-03-30:
> Problem still seems the same, as appears: `Sketch uses 1357754 bytes (**103%**) of program storage space. Maximum is 1310720 bytes.`
>
> **Sketch too big**; see [...]

Will try tomorrow. Today I forgot to bring my ESP32 module from work, but will try tomorrow.
I tried yesterday with Arduino IDE and it compiled up to a point where it errored out but because of an "unused variable".
I'll let you know the exact error I got with Arduino IDE when I have my ESP32 module at home with me!
Thanks

PsySc0rpi0n, 2023-03-31:
Ok, by commenting that line you mentioned I was able to compile and upload the sketch to the ESP32. It is running but when I try to access to the AP, it is asking for a password. In your video you alreayd have the network saved so you don't need to enter any password. But I can't guess the password, right? :p ... What would it be?

BitMaker-hub, 2023-03-31:
The current AP is **NerdMinerAP** and password: **MineYourCoins**

PsySc0rpi0n, 2023-03-31:
@BitMaker-hub I'm sorry if I shouldn't post this here butI dind't want to start another issue as I don't know if this is an issue:

I was able to enter the config panel and enter my home network details but in serial I see this:
```
Config IP Address: 192.168.4.1

*wm:Starting Web Portal 

*wm:26 networks found

*wm:Connecting to NEW AP: MyHomeNetwork(omitted)

*wm:Connect to new AP [SUCCESS] 

*wm:Got IP Address: 

*wm:192.168.1.159 

Should save config

*wm:[ERROR] disconnect configportal - softAPdisconnect FAILED 

*wm:config portal exiting 

WiFi connected

IP address: 192.168.1.159

WiFi connected

IP address: 192.168.1.159

PoolString: solo.ckpool.org

portNumber: 3333

btcString: bc1q....................................

Saving configuration...

{

  "poolString": "solo.ckpool.org",

  "portNumber": 3333,

  "btcString": "bc1......................."

}/n****************************

---> Run monitor Task started

-------------------------

---> Run worker Task started 

[runMonitor Task] -> Printing results on screen 

>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
Button Pressed

Sending  : {"id": 1, "method": "mining.subscribe", "params": []}

Receiving: 

sub_details: 

extranonce1: 

extranonce2_size: 0

method: 

>>>>>>>>> Worker aborted

Sending  : {"id": 1, "method": "mining.subscribe", "params": []}

Button Held

Erasing Config, restarting
...
```

BitMaker-hub, 2023-04-01:
The code check if you hold IO14 (button on TTGO Tdisplay) to reset config. You should comment this line too cause I don't knnow if your board is using it as other function.

PsySc0rpi0n, 2023-04-01:
> The code check if you hold IO14 (button on TTGO Tdisplay) to reset config. You should comment this line too cause I don't knnow if your board is using it as other function.

Could you provide a hint on where that line will be? Which file?

BitMaker-hub, 2023-04-01:
Comment this line :)

https://github.com/BitMaker-hub/NerdMiner_v2/blob/6ecc556966b83bd73a6750a15701a501e9e0d47d/wManager.cpp#L288

PsySc0rpi0n, 2023-04-01:
> Comment this line :)
> 
> https://github.com/BitMaker-hub/NerdMiner_v2/blob/6ecc556966b83bd73a6750a15701a501e9e0d47d/wManager.cpp#L288

Unfortunatelly, commenting that line, seems to put the code execution on hold somehwere:
```
Welcome to minicom 2.8

OPTIONS: I18n 
Port /dev/ttyUSB0, 16:01:20

Press CTRL-A Z for help on special keys

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:1216

ho 0 tail 12 room 4

load:0x40078000,len:10944

load:0x40080400,len:6388

entry 0x400806b4

NerdMinerv2 started......

_
```

PsySc0rpi0n, 2023-04-01:
Ok, @BitMaker-hub I was able to make it work, but it works only for a few seconds, then it crashes and reboot the ESP32.
Not sure if you want or even have the tinme to look at this output. If you see this is way to complex to be helping me, just let me know and I'll quit:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core 0 register dump:

PC      : 0x400d6384  PS      : 0x00060530  A0      : 0x800d644c  A1      : 0x3ffdbba0  

A2      : 0x00000000  A3      : 0x3ffdbbef  A4      : 0x00000001  A5      : 0x00000001  

A6      : 0x00000000  A7      : 0x3ffbbb60  A8      : 0x00000000  A9      : 0x00000000  

A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000050  A13     : 0x3ffca530  

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  

EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d6384:0x3ffdbba0 0x400d6449:0x3ffdbbc0 0x4017647a:0x3ffdbbe0 0x400f9cd2:0x3ffdbc10 0x400f9d6d:0x3ffdbc30 0x400d2db1:0x3ffdbc50 0x40089aca:0x3ffdc040
```

BitMaker-hub, 2023-04-01:
I'm sorry to see you having so many problems. I'll only suggest you at last to try first version of miner. 
There you just need to change config files and is prepared for any ESP32 standard board.

You can check it here [(https://github.com/BitMaker-hub/ESP32_NerdMiner)]

PsySc0rpi0n, 2023-04-01:
> I'm sorry to see you having so many problems. I'll only suggest you at last to try first version of miner. There you just need to change config files and is prepared for any ESP32 standard board.
> 
> You can check it here [(https://github.com/BitMaker-hub/ESP32_NerdMiner)]

Before I try the version you suggest, seems it is running now without crashing. I id nothing different. I ha to leave my house for a while and now I came back and I see this in the serial port:
```
>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
[runMonitor Task] -> Printing results on screen 

>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
[runMonitor Task] -> Printing results on screen 

>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
[runMonitor Task] -> Printing results on screen 

>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
[runMonitor Task] -> Printing results on screen 

>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
[runMonitor Task] -> Printing results on screen 

>>> Completed 0 share(s), 0 hashes, avg. hashrate 0.000 KH/s
```
over and over again! This is already something, no?