BitMaker-hub/NerdMiner_v2 #3: Wifi failure

Wall-8, 2023-04-08:
Hi! Cool project!
When I connect to wifi it reboots multiple time and it connects succesfully to the wifi but ends up disconnected. When I login wifimanager says authentication failure. Tried to find solutions in wifimanager github but no success. Any tips?

BitMaker-hub, 2023-04-10:
Hi, this is something difficult to resolve as WiFiManager library is really simple to use but also closed to make changes on it. 
I experimented same behaviour at the begining but this was related to not finding AP, so only when having really bad signal coverage.

Try rebooting wifi credentials (pressing during 5 seconds top right button) and starting again your module. 
Try accessing AP with other phone

If this fails I'll consider to add another way to configure module

Wall-8, 2023-04-13:
Have rebooted the wifi credentials and accessed with another device. Also stood next to my AP :)
It reboots and loops several time (during the reboot I see connection in the wifi logs and see loaded templates on the screen) but ends up with a an authentication failure.

BitMaker-hub, 2023-04-15:
Updated new realease that helps with reboots. 
Check if you added your BTC address too

closed 2023-04-15 by BitMaker-hub