BitMaker-hub/NerdMiner_v2 #6: If AutoConnect: FAILS during startup a wifi Acces Point is created

BitMaker-hub, 2023-04-18:


Yac79, 2023-04-21:
I just encountered this problem, how do I go about solving it?
[image]

chmeulblu, 2023-05-01:
Hi,
i have sort of that issue too.

when couldn't find the programmed wifi ( for example for running it elsewhere or not on your phone's access point) it does go to the "AP" mode. great . and you can see the APminer wifi available on the phone.
But the AP does not open on my phone when you select that wifi liked it did the first time i programmed the wifi... so impossible to change wifi.

something wrong on my phone ? ( sorry if so to have put that issue here)
Chmeulblu

BitMaker-hub, 2023-05-22:
This was solved on release 1.4