BitMaker-hub/NerdMiner_v2 #7: battery ?

chmeulblu, 2023-05-01:
hi,
since the ESP allows it. i plugged a battery on it. 
[image]
i used one like the one on the photo on  another ESP for another project ( the very good PoS ).

the idea is sometimes, for show or demo purposes  or for fun on a desk to have the miner working without any wire. looks nicer and .. mysterious !
not for a long time but an hour or 2 would be nice.

the ESP is 3,7v so it could work on the 3.4v ... and it .. almot does.
when unplugged with the battery in it  it goes on mining allright i think. but the screen stays black. you recover it when plugging again and see that the process is still on and running. ( looks like it anyway)

so it is close. 
what is it missing ? some mA ?  or is there a hardware safety measure ? 
Could some code arrange that ? :  test if battery on and make it work at a "slower" hashrate for example ? 
or add a little electronic ? ( i saw an addon with a resistor to be able to tune the vent speed...)

It looks to me like a nice possible upgrade
i go from afar  but i am quite willing to help work on it if you and/or other people think it is interesting.

but you haven't proposed or spoken about it, i suppose there is a good reason for that...

this is more a proposal with an issue than a real issue though. let me know if this should be posted elsewhere 

Chmeulblu

BitMaker-hub, 2023-05-01:
Hi Chmeublu

I'll set this issue closed as yesterday I made an update related to a know bug on lilygo tDisplay related to 5V and battery. This let's you work with battery

I'll release a new version today that let you work with battery

closed 2023-05-01 by BitMaker-hub

chmeulblu, 2023-05-10:
hi,
it does work. nice ! great job. i am impressed by the miner. 

the battery can easily be glued inside the top cover. and it works nicely. it also gives the miner some weight : it feels like its mining harder :-)
it may not be good to  have it working all the time ( for the battery also that is drained each time) and more usefull to be able to start it when you want ( demos , academic purposes, fun with friends...) . 
The next step would be to modify the 3D case to allow for adding a small and inexpensive on/off standard electronic switch. no programming there. i could try that whithout any disruption to the project . just a nice addon for those who want to add this feature.
example of switch :
[image]
i ve seen it done for exactly that purpose on ben arc bitcoin point of sale .  Pictures down bellow are an example of modification on the 3D base box ( not the cover). doesn't look that difficult, need no screw or anything. it could be added on the main case,just under the cover , on the plug side. 
[image]
[image]
[image]
i am no fusion 360 or cad expert but i could try to work on it as a contribution to your project. if you are interested. if so can you send me the source 3D files to try that ? 
let me know.