BitMaker-hub/NerdMiner_v2 #10: Battery use - need specials specs ?

chmeulblu, 2023-05-02:
HI, 
i see there is another firmware that deals with Battery. thanks.
i tried it right away and get the same behavior as before however :  

If a battery is plugged in and you disconnect the USB power source to use the miner on the go ...the screen goes black. 
when you plug it in again to an usb power source, after some time,  you see that it has gone on mining during the "black out" . time,  shares, Millions hashes  .. have continued and gone up. Great.
so it is working and may be that shouldn't be an issue ? 

Is my 3.7 v  1000mAh battery too weak somehow ?  it fits the 3.4v ESP need ...
can you recommend some or post the one you used that obviously  works fine including the display and can fit within the miner's box ?

Thanks.

BitMaker-hub, 2023-05-04:
Hi,

Did you tried v1.3? 
That version included a mod to solve this issue. I'm not having them more.

Could you verify you are working with it?

chmeulblu, 2023-05-05:
yep. my mistake. works fine now. shouldn't have been opened , sorry !