BitMaker-hub/NerdMiner_v2 #8: Connect to Miner while it's running

MinerGuy62, 2023-05-02:
How do I change the config one it's running?  Browsing to the IP doesn't seem to work

chmeulblu, 2023-05-02:
Hi,
same .

i tried it also. it won't connect to the ESP IP adress  ( i see the esp on the network and therefore can see it's IP)  from a browser while running but i don't see  it is a bug is it ?
however : not be able to change the wifi  is an real issue for me.  that is may be your point also ?

When the Wifi it is configured to use is made anavailable you can connect to the device Wifi AP Miner that reappears after some seconds.
you can then connect to the wifi miner but it doesn't launch the app. however since it is creating a wifi there is a server adress it has created to allow connection to its wifi. for me it is 192.168.4.1 .
When i get there from a browser ( chrome) i access the App config menu and can type in another wifi / password  and save it. The good news stop there because it won't take it into account ( tried 3 different ones).  too bad. 

to change the wifi i had to erase the firmware ( chrome web ESP tool page connecting to the device using USB) and download it again on the device. and start over the configuration.

 i don't know if i am doing somenthing wrong...

BitMaker-hub, 2023-05-04:
OK, let me start from the begining.

1- The device is currently having only one configuration. The first time. 
2- If you want to modify your settings, you have to delete the entire configuration. Hold right top button almost 10 seconds
3- Then you will be able to acces the APpoint again

**New posible Features**
There is no currently an intention to add a webserver while device is running. I don't know how could it affect.
One other interesting thing we could do is to let you modify your settings when you power it on and a key is pressed.

tppsch, 2023-05-08:
Maybe 2 different settings (one for your home wifi and one for your mobile AP) changable over a buttonpress or something.

chmeulblu, 2023-05-08:
The thing is  the buttons  are not within reach when the miner is in its box … are they ? 
So any button solution … will not be very satisfying. 

may be a solution is possible using the touchscreen version of the display ?  
looks like the touchscreen is very promising to help add feature to your miner. 
great job.

Xeroxxx, 2023-05-16:
Why would I change the WiFi or BTC Address frequently? I don't see the need of being able to do that after configure once. This will probably reduce the hashrate.

chmeulblu, 2023-05-16:
hi Xeroxxx, i don't know if that will reduce the hashrate... Bitmaker may have a view on this.

changing the BTC adress is of no interest i agree.

regarding the wifi usefullness :
the nerdsigner is mainly a fun + demonstrative and learning device. to do that it could  from time to time move to various places ( meetups ...) to show it and use it for explanations or fun. That can be done on the phone's wifi.  Fine. A second usefull wifi would be your home wifi when you want it to stay put.
also when the miner is in its case, it is not possible to reset it using the buttons...
So, if the miner was able to memorize 2 or more wifi and when it boots find one of them on its own, I would personnaly consider this an improvement. of course i am very pleased with the way it is now and thankfull for the great work accomplished.

closed 2024-02-06 by BitMaker-hub

sany3001, 2024-02-06:
Topic is closed. But I would rather invest in new miner that I would take with me somewhere.