BitMaker-hub/NerdMiner_v2 #12: New Feature : PWM Fan controlled by CPU Temperature

hallard, 2023-05-03:
Love this project, and my little contribution is following

I've just quick and dirty created little PCB to be able to fit into T-Display-S3 as is

@BitMaker-hub  Once I got it, I'll do a PR to implement this in code (and show CPU Temp on display), but if you want to add this feature before, of course you can :-)

[image]

[image]

[image]

These boards are untested but should works without any issue 

R1 10K 0805
D1 1N4148 
Q1 BSS138 SOT-23

You can order 3 boards (RAW PCB) for less than $2 shipping included at [OSHPARK](https://oshpark.com/shared_projects/iB2ZQZSp)

Have fun

BitMaker-hub, 2023-05-04:
Hi Hal thats a supercool idea, I'll order them and try it. Do you have these gerbers or I'll make my board?
It could be fun to add them to this project

hallard, 2023-05-04:
Sure here below the gerber files

I can do another PR for hardware related, let me know folder name you want to add to place all

BTW I created this PR #15 to start, so far so good (I put a little heatsink)

[image]

btc-ss, 2023-05-07:
@hallard I really like how you added the internal core temp to the display.