BitMaker-hub/NerdMiner_v2 #13: Not showing on Pool Sites

MinerGuy62, 2023-05-04:
I tried ckpool and zsolo and it doesn't show it's hasing but it does show it's working on the screen at 20+KH/s

I reloaded the new bin files today as well.

MinerGuy62, 2023-05-04:
I saw the other closed issue.  However I use the same btc address on luckymonster and also on ckpool with my other USB miners.

BitMaker-hub, 2023-05-04:
Hi Minerguy, we are doing more tests here. But currently this microminer is not able to reach the difficulty treshold and is not seen by pool. The job is being done but can't even be sent to pool

closed 2023-05-04 by BitMaker-hub

MinerGuy62, 2023-05-04:
Why close this?

BitMaker-hub, 2023-05-04:
I can't make that current miner give enough hashrate to overcome the difficulty treshold of actual pools.
So this is not an issue is a impossible mission.

On next future never know if we will create a solo mining pool for nerdminers.

MinerGuy62, 2023-05-04:
That's why I got the lilygo from Amazon so I could return it.  I will keep an eye on the project

chmeulblu, 2023-05-05:
Hi,
Do you know what is ( around) the necessary hashrate to reach an average pool ?

Good idea the nerd pool site !   If we put all of us together can we reach a pool ..( as a  single pool ?)