BitMaker-hub/NerdMiner_v2 #16: Question: How to view worker after connecting to Braiins Pool?

btc-ss, 2023-05-05:
Hi BitMaker,

Is there a way to view my miner (via Braiins Dashboard on web/mobile app) after I connect the to Braiins slush pool?

Thank you.

laststrawman, 2023-05-07:
I would be interested in this as well. Mine never shows up.

cprados, 2023-05-09:
Same issue here: the URL https:/solo.ckpool.org/users/<my_address> throws 404 Not Found. But I see yours already is working, did you change something @laststrawman ?

BitMaker-hub, 2023-05-09:
Hi all, 

NerdMiner has a really low hashrate that don't let us submit any share through pool difficulty umbral.
This make that pool don't know we are mining. But it's doing it.

To be able to show miner work we should have a pool able to configure miner difficulty to 16 bit shares or let it working as it is trying to solve the block.

closed 2023-05-09 by BitMaker-hub

laststrawman, 2023-05-10:
No I didn't get it working I was giving an example of one that was working.
I see from BitMaker's response that ckpool essentially doesn't show
NerdMiner work because the hash is too low.