BitMaker-hub/NerdMiner_v2 #21: Completed setup but not hashing

joelkuhns, 2023-05-09:
I was able to configure the miner but once it went through the setup, the main screen just shows the count for how long the device has been on.  All other items being tracked on the screen are at zero.

joelkuhns, 2023-05-09:
Cleared the config and tried again, same result.

MinerGuy62, 2023-05-09:
Mine shows hashing but still not showing up on the pool with 1.4

btc-ss, 2023-05-09:
@joelkuhns have you tried a different pool just in case?

MinerGuy62, 2023-05-09:
ckpool and zsolo not registering

BitMaker-hub, 2023-05-09:
Please, check your firewall settings on your router. 
- Sometimes they block pool connections
- Recheck your wifi config. We will try to add an screen advise when WIFI is not connected properly

MinerGuy62, 2023-05-09:
I placed the miner into the DMZ and I now see traffic from it.

So even though the pool doesn't show it, is it still solo hashing?

joelkuhns, 2023-05-09:
I changed it to the pool below and it worked, thanks all!  Probably just blocked on port 3333 or something.  Didn't look into it too far.

btc.zsolo.bid

<br class="Apple-interchange-newline">

closed 2023-05-09 by joelkuhns

Lowlirk, 2023-09-02:
Hey guys same problem. No hashing. Tried ckpool; zsolo; slushpool... Nerdminer pool showed hash for a few minutes (after leaving nminer on for 20 minutes or so at zero hash) and then dropped to nothing. Placed miner on DMZ - no dice.

Lowlirk, 2023-10-18:
> Hey guys same problem. No hashing. Tried ckpool; zsolo; slushpool... Nerdminer pool showed hash for a few minutes (after leaving nminer on for 20 minutes or so at zero hash) and then dropped to nothing. Placed miner on DMZ - no dice.

So, this was an issue with my asus router. Turned off "malicious websites" filtering in ai protection.

BitMaker-hub, 2023-10-18:
Great to know!