BitMaker-hub/NerdMiner_v2 #22: Testing Results Thread: Release v1.4

btc-ss, 2023-05-09:
I have upgraded all 3 of my nerdminers to release v1.4. (firmware bin only, as that was the only file updated).

I have one miner pointed to each of the following pools:

- CK pool
- braiins pool
- zSolo pool

All three pools / miners were relatively stable prior to the upgrade.  They would reset maybe every 24hrs +/-.

Now I am seeing the braiins Nerdminer rebooting itself every 8-10mins approx.    

The other two miners appear to work fine so far.  I will continue to run them to check stability.

MinerGuy62, 2023-05-09:
I am  not seeing mine in ckpool or zsolo and using 1.4

BitMaker-hub, 2023-05-09:
Hi Btc-ss -> I'll check this on braiins pool.
Please take into account what I commented so many times. 

You will never see your miner until we could have a pool with less difficulty.

MinerGuy62, 2023-05-09:
But is it still hashing solo on the pool?

btc-ss, 2023-05-09:
> Hi Btc-ss -> I'll check this on braiins pool. Please take into account what I commented so many times.

After 4 hours only the Braiins Nerdminer still reboots and frequently. 

> You will never see your miner until we could have a pool with less difficulty.

Ok cool.

btc-ss, 2023-05-11:
Update

Ck Nerdminer running for 29hrs - no issues.
ZSolo Nerdminer running for 29hrs - no issues.

Braiins Nerdminer restarts every ~1hr of runtime.

btc-ss, 2023-05-11:
Update

After reconfiguring braiins Nerdminer to point to CK pool it has been running fine for 3hrs.

chmeulblu, 2023-05-11:
great job. nice and skilled community working on it. The nerdminer is flying !
but we can agree at least for those 3 solo pools, that it is for demo and technical practice ( which is great) purposes only ?
those "shares" are not taken into account by the pools right ? there are  too low , they are not visible and can't ever be rewarded ( 0 ).

The nerdminer still is a wonderful device. And I am confident we will figure out in time, hopefully soon,  how to make it count and make mining more decentralized.
sorry if i am mistaken. please tell me.

BitMaker-hub, 2023-05-14:
Hi chmeulblu, you are totally great. This is a cool starting point and a device with more educational purposes than anything.
Pools doesn't know we are hashing because we can't achieve it's difficulty so NerdMiner only tries to reach the block on solo. He only sumbit a share if is the winner one.

cprados, 2023-05-15:
So chances are = 0 after all? Or are there any chances the small they may be, like of wining lotery? Thanks for this great and inspiring proyect whatever the answer is.

WantClue, 2023-05-15:
> So chances are = 0 after all? Or are there any chances the small they may be, like of wining lotery? Thanks for this great and inspiring proyect whatever the answer is.

Virtually the chances are 0

cprados, 2023-05-15:
> > So chances are = 0 after all? Or are there any chances the small they may be, like of wining lotery? Thanks for this great and inspiring proyect whatever the answer is.
> 
> Virtually the chances are 0

Hi, 

Thanks for your response. 

I know chances have to be near 0 with such a low hash rate but my doubt is if a nerdminer would be actually working in the unlikely case it was the first minero to find the solution to the problem. 

F.i. if all miners in the network went offline but our nerdminers, would they have the chance to genetate a block vía the node in the pool? Or that would not be the case as they are not actually in comunication with any node as they are not even being recognized by the pool. 

That is the big diference between 0 and "virtually 0".

Regards.

btc-ss, 2023-05-15:
Update

ZSolo Nerdminer has reached 148hrs of runtime but the hash rate is now zero (0.00). It hasn't crashed as I can still see the timer increase every few seconds. Unclear why hashing stopped.

Ck Nerdminer #1 still running fine after 148hrs as well.

Ck Nerdminer #2 (originally pointed at braiins) is now at 118h. Also running well.

Unclear what happened with ZSolo pool.

In pic ZSolo miner at top, ck miner #1 at bottom. 

[image]

btc-ss, 2023-05-18:
It appears this build is pretty stable. One of my miners reached 200hrs. 

The issue with braiins pool is unclear but at least ck and ZSolo work fine.

closed 2023-05-18 by btc-ss

Cj0078, 2023-09-22:
How do i withdraw my block from public pool account?