BitMaker-hub/NerdMiner_v2 #24: No hash rate update on pool statistic

mrkaqz, 2023-05-12:
Hello,

I just started running nerd miner, it is a great project. 
I have try get BTC address from my trezor and put in a config. firstly i run with ckpool for 2 hours and when I check on the address https://solo.ckpool.org/users/MYBTCADDRESS and it show the page not found then I decided to reset the config and try with  zsolo pool and it was run for 10 hours but still no hash rate show up on the page https://zsolo.bid/en/user/btc?address=MYBTCADDRESS

[image]

Not sure this is normal or a bug on the miner.

pocket998, 2023-05-14:
Hello, 

I have the same problem, No statistic.

Thanks for your answer

btc-ss, 2023-05-14:
In other threads, it's mentioned that the hash rate is so small it doesn't show up.

mrkaqz, 2023-05-15:
Thanks for the clerification. let close this issue

closed 2023-05-15 by mrkaqz