BitMaker-hub/NerdMiner_v2 #26: other TTGO esp32 display

4plus17, 2023-05-13:
Does this also works with the 1.14 inch version instead of 1.9 inch display?

I tried but I got no image on display after flashing firmware.

Link to device: https://www.amazon.de/dp/B07VNG9D52?psc=1&ref=ppx_yo2ov_dt_b_product_details
name of device: ICQUANZX T-Display ESP32 WiFi und Bluetooth Modul Entwicklungsboard für Arduino 1.14 Zoll LCD

Essyarf, 2023-05-14:
Same for me with this 1.9 inch equipment, black screen after the flash.
Anybody have a solution?

Link of equipment 
https://a.aliexpress.com/_m0TB60Y

BitMaker-hub, 2023-05-14:
Hi, currently this code can run there but need to change some graphics things and recompile. ESP32S3 has other memory map
I'll try to make a version to include other boards

closed 2023-05-14 by BitMaker-hub