BitMaker-hub/NerdMiner_v2 #28: All metrics stay 0 (except online time)

Githubber002, 2023-05-15:
What to do when 
Bloc templates and 16bits shares and hashes stay on 0?
Already did a reboot.
Online time seems to work.

[image]

gaelbenoit, 2023-05-15:
```
18:57:25.205 > Client not connected, trying to connect...
18:57:25.334 > [2267507][E][WiFiClient.cpp:268] connect(): socket error on fd 48, errno: 104, "Connection reset by peer"
18:57:25.334 > Imposible to connect to : solo.ckpool.org
18:57:26.334 > Client not connected, trying to connect...
18:57:26.338 > >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.000 KH/s
```

AxelHamburch, 2023-05-15:
Same here. It looks like it is a problem with solo.ckpool.org.

SiriusLightning21, 2023-05-15:
> Same here. It looks like it is a problem with solo.ckpool.org.

I think also this is the problem. Now i changed to eu.stratum.slushpool.com and everything is working again.

janrothen, 2023-05-15:
> 

How do you get the console output?

Kassius84, 2023-05-15:
I had the same problem and switched to btc.zsolo.bid:60 7
This pool is currently working fine for me 👍

janrothen, 2023-05-15:
> Having the same problem right now. Running the latest firmware (tag Release1.4).
> 
> All the values are 0 except the time counter (tried solo.ckpool.org, eu.stratum.slushpool.com, eu.stratum.braiins.com).

Now it worked once after I've reset it a couple of times. But after I assembled it in the case and switched it back on, same behaviour. All 0, except the clock.

btc-ss, 2023-05-15:
Now I understand why my ZSolo miner has went to zero. Thanks all

janrothen, 2023-05-16:
What "fixes" it for me is to reboot the router, while the NerdMiner is running.

Then the mining starts.

PhilippWa, 2023-05-16:
I hat a PiHole in my Network and one Blocklist included "Mining Pools". 

Please check also your Blocker-Tools :)

sepiropht, 2023-06-14:
I still have this problem, and I tried all the solutions in this thread, none worked.

btc-ss, 2023-06-15:
> I still have this problem, and I tried all the solutions in this thread, none worked.
> 
> @gaelbenoit how do you debug the card ?

Have you tried connecting to a different wifi network ?

You could even try using your mobile phone as hotspot to see if it works just to rule out a connection issue.

btc-ss, 2023-06-22:
> @btc-ss , yes i try to connect nerdminer to my smartphone wifi hostpost and now it works ! But it's obviously not very convenient. How can i make the nerd miner connect to my home wifi network ?

Now you need to figure out what on your wifi router is blocking your Nerdminer connection. 

You'll have to research your wifi router for similar issues where it blocks access.  Figure out how to make it allow.  Maybe related to specific port access or needs to be opened. That's my guess.  Good luck and at least now you know it works.

xr6623, 2023-06-22:
I think it can only connect to 2.4 GHz wifi networks, so you may need to enable 2.4 GHz on your wifi router.

sepiropht, 2023-06-28:
> I think it can only connect to 2.4 GHz wifi networks, so you may need to enable 2.4 GHz on your wifi router.

I have both  of them on my router

closed 2024-02-06 by BitMaker-hub