BitMaker-hub/NerdMiner_v2 #32: Does the miner really work?

Roalkege, 2023-05-20:
I have a own BTC Node and setup a pool on it with the mining diff 0.00001.

The Nerdminer is not able to find share with this diff. I find that a bit strange.
Now I tried a CPU miner on my PC that has only one thread (11-18 KH/s) and I find 5-10 shares per second with it. My NerdMiner has 21 KH/s.

alcar21, 2023-05-20:
Yes but the cpuminer don't have implemented the midstate and it's slower than nerdminer.

Nerdminer(ESP32) can reach up to 80khs with the two threads but in this case we be able to only one thread for mining

BitMaker-hub, 2023-05-20:
Hi Roalkege you are right. Currently NerdMiner is not submitting works based on current pool difficulty. It only tries to reach the block using the nbits param of stratum protocol. We directly omitted this part on code because we were unable to reach any pool difficulty.

But it could be really interesting to really try this and could be the first step to setup a NerdMiner pool (just for fun).
Could you be able to reach me on twitter or telegram to start a conversation about this and then to know your setup scheme?
Are you using custom ckpool?

fti7, 2023-05-24:
> Yes but the cpuminer don't have implemented the midstate and it's slower than nerdminer.
> 
> Nerdminer(ESP32) can reach up to 80khs with the two threads but in this case we be able to only one thread for mining
i've also seen that currently only one Mining Thread is active (mining.h:#define THREADS 1)

Whats the exact reason for this? Theoretically there could be two independend Miner Threads or whats the issue here?

BitMaker-hub, 2023-07-24:
Miner is optimized to use 2 threads and fixed some issues with merkle root calculations. Now can work properly good with any low share pool

closed 2023-07-24 by BitMaker-hub