BitMaker-hub/NerdMiner_v2 #36: lost WLAN connection after same hours

TGM3D, 2023-05-23:
Hi, after about 2-4 hours, the miner is currently leaving my WLAN. No new blocks are then requested and the hash rate goes to zero. Everything else seems to continue to work. At the moment I can't see why it should be. Version is the latest 1.4

pakitron, 2023-05-23:
I´m having the same issue. Same version. In my case, in some cases, some minutes after, NerdMiner recconnect to my wifi network; in other cases, I need to reboot (power off and power on). During disconnected periods hash rate go to zero. I have the NerdMiner device 2 meters length to my wifi access point (wifi signal is very strong).

pakitron, 2023-05-24:
Hi, after review other issues for this project and execute some test in my "nerd miner" device, I can conclude that this issue is the same that issue "All metrics stay 0 (except online time) #28"
The root-cause of the problem is some problem related with connectivity with the pool solo.ckpool.org; the device is not able to connect with this pool, receive a connection refuse and the client disconnect of wlan.
I change the server pool (i setup my device with this pool URL: btc.zsolo.bid) and everithing works fine and the device doesn´t disconnect from wifi network anymore.
This issue must be merge with issue "All metrics stay 0 (except online time) #28"

TGM3D, 2023-05-24:
Hi, thanks for answare. I would like to add that the problem no longer always occurs after several resets. I currently have 36 h uptime with the mentioned ckpool.

btc-ss, 2023-05-27:
The longest streak I got up to was 204hrs run time, then something causes the Nerdminer to reboot.  I wonder if something from the pool triggers it.

BitMaker-hub, 2023-06-12:
This should be solved on new versions

closed 2023-06-12 by BitMaker-hub