BitMaker-hub/NerdMiner_v2 #37: Installation help

dackdel, 2023-05-24:
hey how does one flash the firmware, or compile first and then flash the firmware I guess. Is there a blogpost or tutorial beyond "Load the firmware with the binaries from the src/bin folder.
Plug your board and select each file from src/bin with its address"
Cause thats where I am stuck. How do I load the firmware from either the src folder or bin folder. Or from both folders?

dackdel, 2023-05-24:
Ok found one https://www.youtube.com/watch?v=z5GeXZlM4pU

dackdel, 2023-05-24:
Ah shit. I have bought the amoled display https://www.lilygo.cc/products/t-display-s3-amoled https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED I guess that is why I cannot see anything on my screen.

BitMaker-hub, 2023-05-26:
This board uses different screen pins. In future could be supported but not yet

closed 2023-05-26 by BitMaker-hub

dackdel, 2023-05-27:
Can you give me a hint of how i might make it work?