BitMaker-hub/NerdMiner_v2 #44: Error: MD5 of file does not match data in flash!

vendelin007, 2023-05-29:
[image]

BitMaker-hub, 2023-06-08:
This is really strange, should say I don't know why happened. Did you finished the upload?

willinilly075, 2023-06-09:
Experiencing exactly the same issue.
[image]

willinilly075, 2023-06-09:
Any idea what might be causing the hash mismatch?

vendelin007, 2023-06-12:
> This is really strange, should say I don't know why happened. Did you finished the upload?

Unfortunately upload did not finished yet. I already tried many ways. It always end like this. I'm sad ...

BitMaker-hub, 2023-06-12:
Please, try the following:
1- Change your cable for a better one
2- Try changing cable orientation, some don't work proper with it
3- Decrease baudrate
4- Change PC
5- Use other webbrowser
6- Use other linux or windows dedicated tools

willinilly075, 2023-06-12:
Figured this out. It appears Windows Defender was somehow applying file restrictions. Had to right-click every file and remove those. Then it worked out fine.

vendelin007, 2023-06-13:
> Figured this out. It appears Windows Defender was somehow applying file restrictions. Had to right-click every file and remove those. Then it worked out fine.

Did not work for me :-/

vendelin007, 2023-07-21:
New flasher fix the issue. Now it's working :) Thanks.

closed 2024-02-06 by BitMaker-hub