BitMaker-hub/NerdMiner_v2 #50: NerdMiner crashes

tbi88, 2023-06-02:
Hi

My NerdMiner crashes quite often. The longest runtime I have seen so far was a bit more than an hour. But it could also happen after a few minutes. This is what I see on the serial console.

```
Receiving: {"jsonrpc":"2.0","method":"mining.set_difficulty","params":[8000.0],"id":null}
  Receiving: {"jsonrpc":"2.0","method":"mining.notify","params":["00001cba","0f3ec3e037547d1b020e7d9bc424f67c05be412f030478220000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff140303230c04120b7a6400","01380000000001462c4825000000001976a91479d1987f15f5dbd6a3a0c53e4b1303fcee5fe4ef88ac00000000",["3022ea0180125737b2669149658ac05f8388948e2e4c935892fda6e2280f5204","3db41e03f608274c8088b7c63791c7ac20b9b29d84f9a2869387fb8cb46e3289","ce6c8f871c41b959e70bbd0d4188c9133bdcb78a414473c1335e6fa47b71d53b","2d89862b2b7ce5265ab6b32652dfdf33cf2075b2907e0eac538a35f68b4e074a","f022727e3313d13681a38a382553b17d17f42c4611e84597f2b8eb89fdb1f8db","85f658a477ca4708459dad82bd4de86d91e48720eecbecf049642417a4339eaf","e1d8acad8289e94c155d24e78bd70c36fe92edd1c7a02960949d4ba1ef0563c6"],"20000000","18055051","647a0b12",false],"id":null}
  Receiving: {"jsonrpc":"2.0","method":"mining.notify","params":["00001cbb","0f3ec3e037547d1b020e7d9bc424f67c05be412f030478220000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff140303230c041c0b7a6400","01380000000001462c4825000000001976a91479d1987f15f5dbd6a3a0c53e4b1303fcee5fe4ef88ac00000000",["3022ea0180125737b2669149658ac05f8388948e2e4c935892fda6e2280f5204","3db41e03f608274c8088b7c63791c7ac20b9b29d84f9a2869387fb8cb46e3289","ce6c8f871c41b959e70bbd0d4188c9133bdcb78a414473c1335e6fa47b71d53b","2d89862b2b7ce5265ab6b32652dfdf33cf2075b2907e0eac538a35f68b4e074a","f022727e3313d13681a38a382553b17d17f42c4611e84597f2b8eb89fdb1f8db","85f658a477ca4708459dad82bd4de86d91e48720eecbecf049642417a4339eaf","e1d8acad8289e94c155d24e78bd70c36fe92edd1c7a02960949d4ba1ef0563c6"],"20000000","18055051","647a0b1c",false],"id":null}
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: BREAK instr
Core  1 register dump:
PC      : 0x403743c0  PS      : 0x00060436  A0      : 0x00000000  A1      : 0x3fcebef0
A2      : 0x3fc9cb7c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x82028748  A9      : 0x00000003
A10     : 0x3fca6a08  A11     : 0x00060023  A12     : 0x00000003  A13     : 0x82028748
A14     : 0x3fcebea0  A15     : 0x57f3eb07  SAR     : 0x13a2f05e  EXCCAUSE: 0x00000001
EXCVADDR: 0xffffffe0  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x403743bd:0x3fcebef0

ELF file SHA256: 6d4e665f53b1e0e3

Rebooting...```

Any idea what is causing that?

Regards

tbi88, 2023-06-02:
I have erased the flash completely and re-written firmware. Does not help here...

santitf, 2023-06-02:
Hi, I received today my ESP32 s3 from the liligo store. 
Same problem with crashes, same time about an hour max before crash. 
I’m trying differents cables or power adaptors.

santitf, 2023-06-02:
News: I'm using zsolopool under port 6057, but switching ckpool  no crashes for 4 hours.
I see that zsolo has a "very low difficulty" using port 6056 (I will test it).

BitMaker-hub, 2023-06-03:
Hi, I'm going to release an update recently that will solve most of this problems. There were related to Json deserealitzation. So this will no occur more.

santitf, 2023-06-03:
Using zsolo pool with very low difficulty port 6056 do the same crash in an hour aprox.
CKpool seams to be stable.

Jir8taiwan, 2023-06-08:
In my crash case, if I use solo mining such as zsolo and ckpool, it will keep going without problem.
however, when it is doing pool mining such as nicehash, binance or viabtc, it keeps crash loop and disconnect immediately.
I do not able to trace and figure out problem yet, but I try to leave this trouble for reference.
If someone has similar issue or helps, please do some advise.
Thanks a lot.

santitf, 2023-06-08:
> HI all, It could be interesting to do some follow up with new beta version. Where stratum and Json parsing where improved.
> Can you check and give feedback?

Of course!

tbi88, 2023-06-09:
I updated to 1.5 beta yesterday evening. Since then it is running a bit less than 12 hours without an issue. Looks definitely better to me! 

Thanks.

santitf, 2023-06-09:
With 1.5 beta 18 hours pointing to zsolo using 6056 port (very low difficulty).
Shuting down screen, the temps goes down 5-9 degrees using the lilygo case and 26 celsius on room (awsome!!!)

StarGateMiner, 2023-06-09:
Hi @BitMaker-hub Sadly for myself this beta "causes" crashed when I did not have crash before
Using:
solo.ckpool.org | 3333 | https://solo.ckpool.org/

Should I change pools? or perhaps a hard reformat may work?

StarGateMiner, 2023-06-09:
> I updated to 1.5 beta yesterday evening. Since then it is running a bit less than 12 hours without an issue. Looks definitely better to me!
> 
> Thanks.

Which pool?

santitf, 2023-06-09:
> Hi @BitMaker-hub Sadly for myself this beta "causes" crashed when I did not have crash before Using: solo.ckpool.org | 3333 | https://solo.ckpool.org/
> 
> Should I change pools? or perhaps a hard reformat may work?

Hi, I see a weird behavor with hash (go to zero, and up for a few seconds and return to zero) when the miner is connected to the PC after use console with esp tool. With 1.4 this problem no occurs. Needs power off/on to work correctly.

tbi88, 2023-06-09:
> 

btc.zsolo.bid:6057

santitf, 2023-06-09:
> @StarGateMiner
> 
> btc.zsolo.bid:6057

I'm using por 6056. How can we see any difference using 6057 or 6056?

StarGateMiner, 2023-06-09:
> > @StarGateMiner
> > btc.zsolo.bid:6057
> 
> I'm using port 6056. How can we see any difference using 6057 or 6056?

I switched to Zsolo Pool and monitoring to see how things go, but weird thing I see no info in zsolo website of my hashrate stats, wondering if it will not display anything, anyone able to see stats on the site?

santitf, 2023-06-09:
> no

I can't see info too, in zsolo website. 
Is there a pool that sent info about hashrate doing solo mining?

santitf, 2023-06-09:
> I switched to Zsolo Pool and monitoring to see how things go, but weird thing I see no info in zsolo website of my hashrate stats, wondering if it will not display anything, anyone able to see stats [...]

I can't see info too, in zsolo website. 
Is there a pool that sent info about hashrate doing solo mining?

StarGateMiner, 2023-06-09:
I ran now 45minutes on btc.zsolo.bid:6057 and miner froze, no changes happening on screen, needed to restart it

santitf, 2023-06-09:
> I ran now 45minutes on btc.zsolo.bid:6057 and miner froze, no changes happening on screen, needed to restart it

Try port 6056, mine is mining for 16h

santitf, 2023-06-09:
I try ViaBTC solo mining: can see my worker at web pool but affer a few minutes change to online to offlne (ViaBTC dashboard), but the nerd shows hashrate...

StarGateMiner, 2023-06-09:
> I try ViaBTC solo mining: can see my worker at web pool but after a few minutes change from online to offlne (ViaBTC dashboard), but the nerd shows hashrate...

This is the reason I asked, because I want to prove in a way my miner is actually working when it comes to connection with the pool, do not wish to have false hope :-( is there a way to prove the miner is on a solo pool and working fully?

BitMaker-hub, 2023-06-09:
Hi guys, I'll test with this pools but the problem is allways the same.
Miner is getting stratum notify from pool and mines. 
It actualizes everytime a new job is send by pool, but what happen> 
Pool expects an answer every 15 seconds more or less due to its difficulty. 

If they see no answers or no shares sent they cut connection. 

There is also a "keepAlive" method I implemented to try holding connection more. 

But even pool disconnects socket, after NM detecting it reconnects.

THis will be solved with a propietary pool with low low low difficulty share

BitMaker-hub, 2023-06-09:
> ran now 45minutes on btc.zsolo.bid:6057 and miner froze, no changes happening on screen, needed to restart it

Hi, when did you loaded new firmware? are you using 1.5.1 or 1.5?

StarGateMiner, 2023-06-09:
> > ran now 45minutes on btc.zsolo.bid:6057 and miner froze, no changes happening on screen, needed to restart it
> 
> Hi, when did you loaded new firmware? are you using 1.5.1 or 1.5?

My bad, I was using 1.5, upgraded now to 1.5.1 - will retry and monitor firstly on solo pool as this worked well for me originally..

santitf, 2023-06-09:
> THis will be solved with a propietary pool with low low low difficulty share

Are you creating a nerdpool, isn't it?

StarGateMiner, 2023-06-09:
Just checking in again, ok so with firmware 1.5 Im getting miner freezing, same setup was fine with 1.0 and went for days on the 3333 | https://solo.ckpool.org/ - so not sure whats happened :-(

tbi88, 2023-06-09:
Crazy! Mine is running since > 24 hours with btc.zsolo.bid:6057 with v1.5.1. Before with v1.4.0 I never had anything more than two hours.

Are we using the same hardware revisions?

willinilly075, 2023-06-09:
@santitf Are you typing the "clean" url from zsolo (btc.zsolo.bid) or the full stratum link (stratum+tcp://btc.zsolo.bid:6056)?

Can't get mine to work, always stuck at 0.00 hash rate.

santitf, 2023-06-09:
> @santitf Are you typing the "clean" url from zsolo (btc.zsolo.bid) or the full stratum link (stratum+tcp://btc.zsolo.bid:6056)?
> 
> 
> 
> Can't get mine to work, always stuck at 0.00 hash rate.

I type the clean url and the port in the line below.

willinilly075, 2023-06-09:
I got two boards and trying out different combinations of configs.

So far I tried:
- 1.5.1 with both zsolo 6056 and 6057
- Same with 1.5.0
- 1.5.1 with cksolo

In all cases, the result is 0s across the board.

StarGateMiner, 2023-06-10:
> Hi guys, I'll test with this pools but the problem is allways the same. Miner is getting stratum notify from pool and mines. It actualizes everytime a new job is send by pool, but what happen> Pool [...]

I tried version 1.5.1 with different cable also, but issue is same for me on different pools, I get freezing problems where screen freeze and need to restart myself. How do I do a "Hard Reset" of the the device and try and reflash latest code, if issue still exists then something is up with the beta and I will go back to 1.4?

StarGateMiner, 2023-06-10:
> I got two boards and trying out different combinations of configs.
> 
> So far I tried:
> 
>     * 1.5.1 with both zsolo 6056 and 6057
> 
>     * Same with 1.5.0
> 
>     * 1.5.1 with cksolo
> 
> 
> In all cases, the result is 0s across the board, even though ports are open.
> 
> [image]

Thats a good point, Im also using Asus router, you saying we need to open ports on router, I did a port check and can see port 3333 on my device is closed?

santitf, 2023-06-10:
> I got two boards and trying out different combinations of configs.
> 
> So far I tried:
> 
> * 1.5.1 with both zsolo 6056 and 6057
> * Same with 1.5.0
> * 1.5.1 with cksolo
> 
> In all cases, the result is 0s across the board, even though ports are open.
> 
> [image]

Hi! I never open ports for mining.
I have an asus router too. Please, check if AiProtection - Two-Way IPS is disable.

willinilly075, 2023-06-10:
> I have an asus router too. Please, check if AiProtection - Two-Way IPS is disable.

This seems to have fixed the issue! Thank you!

Also noticed the DNS Server set up on the router might be blocking certain domains associated with crypto mining. I myself use NextDNS, but switching to something mainstream like Google DNS allowed the miner to work for a couple of hours. I whitelisted zsolo.bid and ckpool.org, and now I can use NextDNS again.

StarGateMiner, 2023-06-11:
> > Hi, I'm going to release an update recently that will solve most of this problems. There were related to Json deserealitzation. So this will no occur more.
> 
> I'm testing the beta version and got many times:
> 
> ```
> >>> Completed 0 share(s), 36475 Khashes, avg. hashrate 22.98 KH/s
> Received message from pool
> Receiving: {"result":true,"error":null,"id":6}
> Parsed JSON: unknown
> >>> Completed 0 share(s), 36491 Khashes, avg. hashrate 21.12 KH/s
> ```
> 
> It will continue on.

Which pool you connecting to? can you try another pool?

StarGateMiner, 2023-06-11:
I followed this guide: https://github.com/BitMaker-hub/NerdMiner_v2/issues/59#issuecomment-1585710679
Seems to have helped with the freeze issue, I think I will flash this way going forward in the future.
Been on btc.zsolo.bid with port 6056 due to "Very Low Difficulty") now and no crash or freeze for over 45min where before it happened.

santitf, 2023-06-11:
> > > > Hi, I'm going to release an update recently that will solve most of this problems. There were related to Json deserealitzation. So this will no occur more.
> 
> > > 
> 
> > > 
> 
> > > I'm testing the beta version and got many times:
> 
> > > ```
> 
> > > >>> Completed 0 share(s), 36475 Khashes, avg. hashrate 22.98 KH/s
> 
> > > Received message from pool
> 
> > > Receiving: {"result":true,"error":null,"id":6}
> 
> > > Parsed JSON: unknown
> 
> > > >>> Completed 0 share(s), 36491 Khashes, avg. hashrate 21.12 KH/s
> 
> > > ```
> 
> > > 
> 
> > > It will continue on.
> 
> > 
> 
> > Which pool you connecting to? can you try another pool?
> 
> 
> 
> btc.zsolo.bid:6056
> 
> It's not crashing as mentioned in this issue, but in the logs I see while I monitor it parsed the json and it doesn't know what the message from the pool was.

I see the same logs like you, using zsolo.

StarGateMiner, 2023-06-11:
> btc.zsolo.bid:6056 It's not crashing as mentioned in this issue, but in the logs I see while I monitor it parsed the json and it doesn't know what the message from the pool was.

Noob question, how do I capture/see the logs?

StarGateMiner, 2023-06-13:
> Hi
>
> My NerdMiner crashes quite often. The longest runtime I have seen so far was a bit more than an hour. But it could also happen after a few minutes. This is what I see on the serial console.
>
> ``` [...]

Noob question, how do I capture/see the logs?

munsterlander, 2023-11-20:
Just got a brand new unit from the store with everything pre-installed.  It will not stay up for longer than 8 minutes, then it crashes and reboots.  The ESP Tool produces this console output:

```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: ec:da:3b:95:36:f4
Uploading stub...
Running stub...
Error: Timeout
```

deyw-bit, 2023-11-20:
Facing the same issue as described by munsterlander, with similar console output. 
Testing new preinstalled device. The device connects to WiFi, switching screen works fine, it displays correctly the current block, current time, correct global hashrate, etc. 
BUT it doesn't mine. 
I have tried to change the configuration to various mining pool, but no success. 
On the mining screen it shows: Block Templates 0, Best difficulty 0.0000 ... 0.00 KH/s.
I flashed it successfully with https://flasher.bitronics.store/ -- Successfully flashed NerdMinerv2 1.6.2 original board T-Display S3. 
But the issue persists. The device keeps restarting after approx 12 to 15 minutes. 

Console output: 

`Client: connected>true / subscribed>false / wificonnected>true
                                                                                                                                                                                                                                                                                                                                              >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                                                                                                                                                                                                                                                                                                          >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                                                       >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                                                                   >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                                                                                                                                                                                                                              >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                                                                                                                                                                                                                                          >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                       >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                   >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                                                                                                                                                              >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                                                                                                                                                                          >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                                                                                                                                                                                                                                                                                                                                     >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                   >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                                                                                              >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                                                                                                          >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                                                                                                                                                                                                                                                                     >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                                                                                                                                             [... cut ...]                                   >>> [i] Miner: newJob>false / inRun>false) - Client: connected>true / subscribed>false / wificonnected>true
                                                                              [WORKER] ==> Mining subscribe
                                                                                                             Sending  : {"id": 1, "method": "mining.subscribe", "params": ["NerdMinerV2"]}

>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
`

munsterlander, 2023-11-21:
Yup, basically seems like these are crapware now. I cannot get it to mine at all. Tried different pools, flashed, everything.

deyw-bit, 2023-11-21:
Created a new [issue](https://github.com/BitMaker-hub/NerdMiner_v2/issues/234#issue-2003995950) for the problem related to not getting any block templates (device not able to connect to mining pool).

closed 2024-06-30 by BitMaker-hub