BitMaker-hub/NerdMiner_v2 #60: Not an issue a question

Githubber002, 2023-06-10:
Not an issue, but a question. Don't know where else to place it:

Can someone explain this that is in the readme? What does it mean it is not seen by the mining pools?

IMPORTANT Miner is not seen by pools due to its low share difficulty. You can check miner work seeing logs via UART or mainly setting up your own pool with a low difficulty rate as 1e-9

Can someone explain if there will come a mining pool just for nerdminers and why that would be better?

Githubber002, 2023-06-11:
THank you.

Am I correct if I write that if a mining pool cannot see the nerdminder they also cannot give the reward (when applicable)?

santitf, 2023-06-11:
> THank you.
> 
> Am I correct if I write that if a mining pool cannot see the nerdminder they also cannot give the reward (when applicable)?

Hi,
I copy a response from ckpool forum to the same question: 

"With only 20 kh/s you will need to wait a lot (some days or weeks) to submit at least 1 valid share 

And without valid share submitted and accepted, the pool can't show any stats"

closed 2024-02-06 by BitMaker-hub