BitMaker-hub/NerdMiner_v2 #63: Zsolo pool doesn't works

santitf, 2023-06-17:
My 3 devices did go to zero 9 hours ago (zsolo and port 6056). 
I tried to reset or reboot my router with no luck. Change to ckpool fix the problem.
I see that in zsolo web there isn't mention to the very low hash port (dissapear) and only normal diff (>500)
Using 6057 doesn't works anymore.
Back to ckpool before nerdpool will go online. ;)

Console : btc.zsolo.biz
Resolved DNS got: 0.0.0.0
                         >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                     >>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true
                                                                         [WORKER] ==> Mining subscribe
                                                                                                        Sending  : {"id": 1, "meth>>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                      >>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true
                                                          >>> Completed 0 share(s), 0 Khashes, avg. hashrate 0.00 KH/s
                                                                                                                      >>> [i] Miner: newJob>false / inRun>false) - Client: connected>false / subscribed>false / wificonnected>true
                                                                                                          Client not connected, trying to connect...
>>> Completed 0 share(s), 0 Khashes, avg.

StarGateMiner, 2023-06-17:
I noticed this last night also, looks like they removed from their website, wonder if there is explanation somewhere, this seems to be problem with pool provider I guess. I will stick with ckpool

cyrix000, 2023-06-17:
yes it's a zsolo Problem, i Point a bigger miner on it, Same Problem.

lawrenz1337, 2023-06-19:
from zSolo support:

[image]

apparently very low are gone forever

closed 2024-02-06 by BitMaker-hub