BitMaker-hub/NerdMiner_v2 #74: solo.ckpool.org

zeus669, 2023-06-21:
It`s not an error. 

But how I can see if it works? 
If I use this _Solo.ckpool.org currently only displays minimal per user and worker information.
To access user statistics, enter your BTC address below:__

If I put my BTC address below an check it - nothing happend (Not Found)

StarGateMiner, 2023-06-21:
Exactly the same issue I have, I raised in another thread but did not get response, how do we know its really mining?

BitMaker-hub, 2023-06-21:
Hi all, I explained this on so many threads. 
Miner is unable to reach any pool difficulty threshold and thats why it doesn't send any share to pool.
Code can be readed and verified. If a share is sent to pool and miner don't meets pool difficulty threshold it returns an error to miner, you can try it.

By now, we started some tests with ultra low diff pool with success. 
We will try to give people the way to connect Nerdminers to this pool when we will finish to build the website.

BitMaker-hub, 2023-06-21:
@lawrenz1337 If you run your own btc node, with a local pool with difficulty about 1e-9 you will see how NerdMiner is sending shares without the need to modify anything on code.

zeus669, 2023-06-21:
@lawrenz1337 

can you descripe it how it works? I also have a local umbrel node

lawrenz1337, 2023-06-21:
@zeus669 if you have umbrel node then you should install btc node there, then configure a pool that will utilize that node that's basically it, not sure if umbrelOS will let you configure a pool on that exact device or not, but give it a shot

zeus669, 2023-06-21:
@lawrenz1337 

Thanks, I think it down`t work with umbrel. I don`t find anything. I think I must wait for @BitMaker-hub for his website

zeus669, 2023-06-22:
How can I see the logs via UART?

BitMaker-hub, 2023-06-26:
You can see logs connecting your nerdminer to any USB of your PC and using any terminal application like Putty or online UART terminal app like https://www.serialterminal.com/

To setup your local pool check ckpool guidelines

closed 2024-02-06 by BitMaker-hub

Arn0m, 2024-09-16:
we are have the same problem....running a full-node and a own solo-pool by ckpool-solo but we can't lower the pool-diff to a smaller diff then 1! Thats really very disgusting. a lot of time to learn, read, understand and then one of the best solo-pool project we can't change the settings as we need it! 

i checked all files at the /src/ specially the ckpool.c and there are the default settings but if we change it we can't lower them!

Has anyone a Idea how to fix it? Public-Pool can do it ... we all understand with small ESP32 Boards there need a tone of luck to get a block but as a hobby we would enjoy the community. 
So back to the question .... how we can modify the settings to make ckpool nerdminer ready?