BitMaker-hub/NerdMiner_v2 #75: Newbie Question: whats a 16 bit share / 32 bit share?

lupinglui76, 2023-06-24:
Hey,

Im new to the mining business (lol).

Can someone explain what  a 16 / 32 bit shrare is? I cant find the information while googeling.

golden-guy, 2023-06-24:
That's the number of leading zeroes in the calculated hash. For instance, in hex notation, the hash value has 64 digits (because a single hex digit is a 4 bit value). Which means that a 16 bit share has 4 leading zeroes and a 32 bit share has 8 leading zeroes.
16 bit share: 0000...
32 bit share: 00000000...

sk3pt1k, 2023-06-24:
The most recent block has a hash of 00000000000000000003c76a8150e35f2996d9e085be60150aac456dc8aef610 - did that make it a "76 bit share" (19x4)?

lupinglui76, 2023-06-24:
Ah okay, thank you for explaining :)

got my first 32 bit after 2 hours 👍 

now waiting since 42 hours for the next :D

lupinglui76, 2023-06-24:
64 bit share would be 16x0?

closed 2023-06-24 by lupinglui76

ANT-Family, 2023-10-03:
thanks, for the answers. didn´t knew also