BitMaker-hub/NerdMiner_v2 #80: Unable to update firmware

Florian-Epitech, 2023-06-30:
Hello,
After deleting the firmware on ESPTools on Program 115200 I wanted to reinstall it, I did clone the repo and put the correct firmware folder but now I have this error and with a constant black screen on my box. What can I do?

[image]

carraresi, 2023-06-30:
Me too. Same error.

Florian-Epitech, 2023-06-30:
I found the error. For me it was the firmware. You need to go in the folder bin/flash_download... and launch the software. Then, just select the version of your NerdMiner (ESP32 S3 in my case). Now select the different files from /bin except the beta firmware and flash it on your Nerd. Reboot him (unplug plug) and it works. 
Tell me if you need help

chrisjmccreadie, 2023-07-03:
Any idas how you would do the flashminer updated tool on a mac?

Florian-Epitech, 2023-07-03:
I don't know if it's possible but try with this https://github.com/BitMaker-hub/NerdMiner_v2/issues/59#issuecomment-1585710679

ryenski, 2023-07-04:
This appears to be an issue with ESP Tool, not with the NerdMiner firmware. I was able to load the firmware by downloading the [esptool-js](https://github.com/espressif/esptool-js) repository and checking out tag [v0.2.2](https://github.com/espressif/esptool-js/releases/tag/v0.2.2). Follow the instructions in the readme to run ESP Tool in dev mode.

BitMaker-hub, 2023-07-08:
Hi guys, 

Seems there is a problem now with current online ESP-tools. You can use this flasher by now

https://valerio-vaccaro.github.io/diyflasher/

hansmade1, 2023-07-11:
I can't program my NM with the ESP Tool or either of the diyflasher links. The ESP Tool goes straight to 'leaving' and there is no response from either diyflasher links.  I have been able to update the firmware before but not now.  I look forward to some detailed instruction to resolve this.

thomashf, 2023-07-17:
I did not manage to flash latest firmware with the ESP onlinetool but the diyflasher worked for me.
Thanks!