BitMaker-hub/NerdMiner_v2 #81: LILYGO T-Display 1.14

rhcpdu13, 2023-07-03:
Hi !
I'm trying to test NerdMiner on my LILYGO® TTGO T-Display 1.14 https://fr.aliexpress.com/item/33048962331.html?gatewayAdapt=Msite2Pc

But the screen is black. The flash is done with the tool flash_download_tool_3.9.3
My first question is : It is compatible ? if yes, how i can do it ?

Thank you !!

ymongo, 2023-07-12:
same here, https://bitmaker-hub.github.io/diyflasher/ says Successfully flashed NerdMinerv2 Latest Version T-DisplayS3

but screen is black, and no WIFI AP

WantClue, 2023-07-13:
> Hi ! I'm trying to test NerdMiner on my LILYGO® TTGO T-Display 1.14 https://fr.aliexpress.com/item/33048962331.html?gatewayAdapt=Msite2Pc
>
> But the screen is black. The flash is done with the tool [...]

You can install this piece of software on almost every esp32, you can achieve this by using the esp32-devkit PlatformIO tasks.

ymongo, 2023-07-14:
> PlatformIO tasks

do you have any step by step on how to PlatformIO? In any case I'm investigating it now, looks exciting!

ymongo, 2023-07-16:
so far: 
- esp32-devkit PlatformIO task on VSCode gives me missing FREETROS.h file...
- TTGO Display task gives me missing sdk file....

AbsoluteDEF, 2023-07-24:
Yeah mine too, but do we need different firmwares for different chips?

BitMaker-hub, 2023-07-24:
We are adding support on this, there is still some incompatibility with libraries I'm holding

ZycoMonk, 2023-08-04:
i have the same problem... 
Chip is ESP32-D0WDQ6 (revision 1)
Crystal is 40MHz

BitMaker-hub, 2023-08-04:
This week started with ESP32devkit support, soon will give TDisplay 1.14 to

ymongo, 2023-09-16:
Working fine here on TDisplay 1.14 , I think issue can be closed!

BitMaker-hub, 2023-09-17:
Thanks! Issue closed

closed 2023-09-17 by BitMaker-hub

Bibit2669, 2023-11-07:
hello, for me it is impossible to detect T-Display 1.14 on the ESPtool. nor even on the .io, how to make it visible to install the .bin?