BitMaker-hub/NerdMiner_v2 #83: Programing ESP32 S3

TrustinDo, 2023-07-08:
I was trying to program my lilygo esp32 when i got done putting the file in and pressing program the esp tool said leaving..
[image]

21orangehat, 2023-07-08:
Hey @TrustinDo ,

I was having the same problem.

I saw that this online platform is having issues with NerdMiner_v2 and the guys moved the flashing step to Platform.IO

They still need to update the REAMD.me but you can follow this[1] video.
[1]https://www.youtube.com/watch?v=cI0fCBFyQok

Best

BitMaker-hub, 2023-07-11:
Hi I added a new flasher utility to github expressed as https://bitmaker-hub.github.io/diyflasher/

closed 2023-07-11 by BitMaker-hub