BitMaker-hub/NerdMiner_v2 #84: support other ESP32 Boards ?

chromstyler, 2023-07-08:
is there any chance to get this to work on a
ESP32-S3 ESP-WROOM-32 Development Board
like this:
[ESP32-S3 ESP-WROOM-32 Development Board](https://www.aliexpress.com/item/1005005716874393.html)

both are ESP-S3 Boards
but if i flash the bin files screen is blank, but the wifi ap is there and the menue works, so maybe only a few modifications needed ?

BitMaker-hub, 2023-07-11:
Hi I'm trying to compile for them too, I'll add them to flasher soon

chromstyler, 2023-07-12:
nice, thank you... will you post a feedback in here when done ?

dackdel, 2023-08-02:
Looking to be able to compile into this https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED board.

BitMaker-hub, 2023-08-03:
Hi, at the end of the week we will provide support for [ESP32-S3 ESP-WROOM-32 Development Board](https://www.aliexpress.com/item/1005005716874393.html)

closed 2023-08-03 by BitMaker-hub

chromstyler, 2023-08-16:
is there a way to get the [ESP32-S3 ESP-WROOM-32 Development Board] (https://www.aliexpress.com/item/1005005716874393.html) working ?