BitMaker-hub/NerdMiner_v2 #90: Access the miner via WLAN after Config ?

chromstyler, 2023-07-12:
Hey there,
is this normal, that after configuration i cant access the miner via its ip address ?
for Updates or change of parameters like pool timezone etc ?

WantClue, 2023-07-13:
Yes, the miner closes its access point mode and switches over to connect to the Wi-Fi that you put in there. 
If you want to make changes, you can simply press the KEY button directly after it boots and you see the initial screen.
It will change to AP mode again, and you can make changes to the config.

chromstyler, 2023-07-13:
ah okay i thought, i could access the miner via wlan and get on a "service" page...

sebrock, 2023-07-13:
> 

And that is exactly what you are doing when you connect to the miner's Access Point when it is in AP mode.
But yeah, you'll have to disconnect, boot, and do as @WantClue said in https://github.com/BitMaker-hub/NerdMiner_v2/issues/90#issuecomment-1633788707

closed 2023-07-24 by BitMaker-hub