BitMaker-hub/NerdMiner_v2 #97: Nerdminer does not submit valid shares

benjamin-wilson, 2023-07-18:
My ultra-low diff compatible pool (https://github.com/benjamin-wilson/public-pool) does not receive valid shares from nerdminers. All other miners tested submit valid shares. Either the header construction or the double sha hashing algorithm appears to be incorrect.

streamofstars, 2023-07-19:
> does not receive valid shares from nerdminers

See https://github.com/BitMaker-hub/NerdMiner_v2/issues/32#issuecomment-1556031121

benjamin-wilson, 2023-07-19:
I should've clarified that it does receive shares but they are below the specified 1e-9 difficulty.
https://github.com/BitMaker-hub/NerdMiner_v2/blob/3ab4a2fb058bbecefa274bd936490ade72a384db/src/mining.cpp#L301C20-L301C20

golden-guy, 2023-07-23:
Should be all fixed now with https://github.com/BitMaker-hub/NerdMiner_v2/pull/100

Thanks for the test sessions to sort out all these issues!

BitMaker-hub, 2023-08-03:
This was solved on 1.6.0

closed 2023-08-03 by BitMaker-hub