BitMaker-hub/NerdMiner_v2 #98: Feature request: Bitcoin puzzle solving mode instead of mining

streamofstars, 2023-07-19:
A feature request: alternative working mode in which the device attempts to solve the next unsolved Bitcoin puzzle from the list. See thread: https://bitcointalk.org/index.php?topic=5218972

BitMaker-hub, 2023-08-04:
What an amazing work mode is this! I have to take a look better but this is a new rabbit hole!

streamofstars, 2023-08-09:
See also the following links, as the one I already posted is just one of many (there are at least three threads about it on bitcointalk forum). The simplest algo would be: take a random number from given puzzle range (or just randomize a starting point within the range and then iterate) and use it as a private key, generate public key, do rmd160(sha256(public key)), compare it to rmd160 extracted from funded address of given puzzle.

https://privatekeys.pw/puzzles/bitcoin-puzzle-tx
https://privatekeyfinder.io/bitcoin-puzzle/
https://bitcointalk.org/index.php?topic=1306983
https://bitcointalk.org/index.php?topic=5166284