BitMaker-hub/NerdMiner_v2 #99: Nerdminer Stops Mining

TrustinDo, 2023-07-20:
After installing the new software my miners would be at 45k/s and then 0 and they back to 45k/s and would repeat that ever few minutes.
[image]

lupinglui76, 2023-07-21:
i got the same problem...

Bergge, 2023-07-28:
I had the same problem where the hashrate fluctuated between 45 KH/s and 0 KH/s. I found read [Issue#78](https://github.com/BitMaker-hub/NerdMiner_v2/issues/78) where a similar problem was discussed and the solution worked for me. I powered my nerdminer through the USB port of my home NAS. After using a "real" charger everything worked fine. So maybe the solution ist to connect your nerdminer to a normal charger and don't use the USB port of your NAS or your router.

ElChopChop, 2023-08-04:
@TrustinDo can you confirm if you've tried a different USB-C cable, preferably plugged directly into a power socket. Other users have reported this working for them.

closed 2024-06-30 by BitMaker-hub