BitMaker-hub/NerdMiner_v2 #104: Stops mining after 2 days

mkellner, 2023-07-25:
Runs well, clock is still running. Cycles through screens by button press - no problem.

But after about 2 days, stops mining. 0.00 KH/s

[image]

Hellangel1987, 2023-07-29:
mine is not starting is your miner starting right after boot up ?

mkellner, 2023-07-29:
Twice it got to two days and some and stopped mining. I recall both of them had 2 32BITs SHARES.

Now, it is still running for 4 days. There are no 32BITs SHARES, but 229035 16 bit shares.

chromstyler, 2023-07-29:
Me the same After updating to lastes fw

Hellangel1987, 2023-07-29:
My wifi passwort was wrong maybe check yours mine is working again no problems

mkellner, 2023-07-31:
[image]

Still running for nearly 6 days. No 32BITs SHARES, so maybe the hang is 32BITs SHARES related. If it hangs again, I'll report.

mkellner, 2023-08-03:
Nope. Up to 9 days then stops mining.
Maybe its at the other end?

[image]

ElChopChop, 2023-08-05:
Someone else has this issue and resolved it by changing power supply. I think they were originally plugged into a NAS for power and plugging directly into a wall socket fixed the issue. What is yours plugged into?

mkellner, 2023-08-05:
5v usb hub, externally powered by 12v wallwart.

The app was still running, just not doing any mining. As if data stopped coming in or something. Network was still running for other machines.

Capswiss, 2023-08-06:
The same applies to me, after about two days the mining stops. Unplugging and plugging back in doesn't help either. No mining again after restart.
The app was still running, just not doing any mining. As if data stopped coming in or something. Network was still running for other machines.

I can still toggle through the three screens.
But every now and then (about every minute) it shows the screen (see picture) for a few seconds...
When I connect to the device, I can see that all of my WiFi and BTC address settings are still there and correct. Also the time code "2" for Europe.
[image]

timglenn123, 2024-04-12:
My device stopped mining after 2 days.. unplugging and trying again.

ozonzo, 2024-06-19:
Hey guy's, I got mine kinda later here, but I have same problem, everything is conneceted, i reset few times but still o.oo KH/s. Show time,price, but even block templates is 0.

sany3001, 2024-06-19:
I was observing no mining on 1.6.3 from time to time. It was some issue with wifi. I recommend to check if AP of miner is not active.

ozonzo, 2024-06-19:
Hello,

Yes, when I connect on the wifi it shows AP is not active, but I don’t know
how activated, changed the cables and outlets and reset many times but
unfortunately no success!

sany3001, 2024-06-20:
I would try wiping the whole flash (maybe with esp tool) and using 1.6.3 or newer FW and configuring all from scratch.

Maybe before doing this you can check debug output (just serial port with 115200, 8N1) on usb port.

closed 2024-06-30 by BitMaker-hub

Vumet3r, 2025-02-09:
I have (had) a plane esp32 wroom 32d and flash it but after 1hour I've got a notification on my computer that the port got disconected because it was using a lot of energy, maybe a shortcut(?) because the board is not responding anymore. I managed to conect it to the wifi. Any hit about what could be the issue? I guess is fried now besides I cannot see any burned component...