BitMaker-hub/NerdMiner_v2 #112: feature request: 3D printable case enhancements

roblatour, 2023-07-30:
First, thank you for making the 3D printable .stl files available, and for your work on this project.

Here is a feature request:

If the 3d printed case was originally designed using Fusion 360 can you please provide the .fd3 files

If not, could you please upgrade the case to have the following features:

1. a way to press the ESP32 T-Display S3's top, bottom and reset buttons when it is in case,

2. a grove to allow the fan wire to enter the case when when it is mounted in such a way that the air is flowing into the case rather than out of it, and 

3. slight modifications around the screen boarders so that the screen fits more properly within the case.

Thank you

ElChopChop, 2023-08-04:
Hey, there's quite a few NerdMiner cases on Thingiverse including this one with access to all buttons:

https://www.thingiverse.com/thing:6053006

The ESP32-S3 tends to be a snug fit, but as long as all plastic tags/imperfections are removed then it should clip in without too much issue. Apply outwards pressure on either end if it's still too tight, or even heat it up first using a heat gun or hairdryer.

You can import STL files into most applications; I've been using NX, Blender and Tinkercad - all allow importing of STL files (appreciate it may not be ideal, but should be enough to add a simple cable channel).

Hope that helps.

roblatour, 2023-08-04:
Thanks.

I use Fusion 360 Community edition, and while it does support importing .stl files, converting them so that you can make modifications to the model is sadly a feature of the pro (paid) version.

While I do like the style of your case for this project as it doesn't have buttons or button holes (I double checked the link you provided above) so its only good if you don't ever need to push the buttons once its up and running.

In any case, don't worry about it, I had previously designed a case for the T-Display S3 which does have holes to push the buttons thru:  https://www.printables.com/model/339237-lilygo-t-display-s3-without-header-pins-case
so I'll just work with that for now.  

If I design another case that supports a fan I'll let you know.

closed 2023-08-04 by roblatour