BitMaker-hub/NerdMiner_v2 #114: sha256 library error

wh1ter0z, 2023-08-01:
hello 

my error is:  

**Multiple libraries were found for "WiFi.h"
mining.cpp:6:10: fatal error: wolfssl/wolfcrypt/sha256.h: No such file or directory
 Used: C:\Users\Rx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi
 #include <wolfssl/wolfcrypt/sha256.h>
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
wolfssl/wolfcrypt/sha256.h: No such file or directory**

when i include **#include <wolfssl.h>** in project 

my error will be :

**C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-master\src\NerdMinerV2\mining.cpp: In function 'void runMiner(void*)':
mining.cpp:244:5: error: 'Sha256' was not declared in this scope
     Sha256 midstate[32];
     ^~~~~~
mining.cpp:247:11: error: expected ';' before 'sha256'
     Sha256 sha256;
           ^~~~~~~
           ;
mining.cpp:250:19: error: 'midstate' was not declared in this scope
     wc_InitSha256(midstate);
                   ^~~~~~~~
C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-master\src\NerdMinerV2\mining.cpp:250:19: note: suggested alternative: 'setstate'
     wc_InitSha256(midstate);
                   ^~~~~~~~
                   setstate
mining.cpp:250:5: error: 'wc_InitSha256' was not declared in this scope
     wc_InitSha256(midstate);
     ^~~~~~~~~~~~~
mining.cpp:251:5: error: 'wc_Sha256Update' was not declared in this scope
     wc_Sha256Update(midstate, mMiner.bytearray_blockheader, 64);
     ^~~~~~~~~~~~~~~
mining.cpp:285:32: error: 'sha256' was not declared in this scope
       wc_Sha256Copy(midstate, &sha256);
                                ^~~~~~
C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-master\src\NerdMinerV2\mining.cpp:285:32: note: suggested alternative: 'shares'
       wc_Sha256Copy(midstate, &sha256);
                                ^~~~~~
                                shares
mining.cpp:285:7: error: 'wc_Sha256Copy' was not declared in this scope
       wc_Sha256Copy(midstate, &sha256);
       ^~~~~~~~~~~~~
mining.cpp:287:7: error: 'wc_Sha256Final' was not declared in this scope
       wc_Sha256Final(&sha256, hash);
       ^~~~~~~~~~~~~~
mining.cpp:364:20: error: 'sha256' was not declared in this scope
     wc_Sha256Free(&sha256);
                    ^~~~~~
C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-master\src\NerdMinerV2\mining.cpp:364:20: note: suggested alternative: 'shares'
     wc_Sha256Free(&sha256);
                    ^~~~~~
                    shares
mining.cpp:364:5: error: 'wc_Sha256Free' was not declared in this scope
     wc_Sha256Free(&sha256);
     ^~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Rx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'Sha256' was not declared in this scope**

please help me

BitMaker-hub, 2023-08-03:
Hi, please, try with last dev branch. 
We are using a custom SHA lib not need more wolfssl

closed 2023-08-03 by BitMaker-hub

wh1ter0z, 2023-08-05:
LOVE YOU BITMAKER

wh1ter0z, 2023-08-05:
c:/users/rx/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\mining.cpp.o:(.literal._Z8runMinerPv+0x64): undefined reference to `nerd_midstate(nerd_sha256*, unsigned char*, unsigned int)'
c:/users/rx/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\mining.cpp.o:(.literal._Z8runMinerPv+0x68): undefined reference to `nerd_double_sha(nerd_sha256*, unsigned char*, unsigned char*)'
c:/users/rx/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\mining.cpp.o: in function `runMiner(void*)':
C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-dev\src\NerdMinerV2/mining.cpp:258: undefined reference to `nerd_midstate(nerd_sha256*, unsigned char*, unsigned int)'
c:/users/rx/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-dev\src\NerdMinerV2/mining.cpp:298: undefined reference to `nerd_double_sha(nerd_sha256*, unsigned char*, unsigned char*)'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Rx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board LOLIN D32.

wh1ter0z, 2023-08-05:
another error