BitMaker-hub/NerdMiner_v2 #120: another error  line 258 and 298

wh1ter0z, 2023-08-05:
another error  line 258 and 298 

runMiner(void*)': C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-dev\src\NerdMinerV2/mining.cpp:258: undefined reference to nerd_midstate(nerd_sha256*, unsigned char*, unsigned int)'

c:/users/rx/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Rx\OneDrive\Desktop\NerdMiner_v2-dev\src\NerdMinerV2/mining.cpp:298: undefined reference to `nerd_double_sha(nerd_sha256*, unsigned char*, unsigned char*)'
collect2.exe: error: ld returned 1 exit status

[image]

_Originally posted by @wh1ter0z in https://github.com/BitMaker-hub/NerdMiner_v2/issues/114#issuecomment-1666476955_

BitMaker-hub, 2023-08-19:
Please use Platdorm IO

closed 2023-08-19 by BitMaker-hub

wh1ter0z, 2023-10-04:
> Please use Platdorm IO

Fine