BitMaker-hub/NerdMiner_v2 #123: Since yesterday all miners stops

drdr34m, 2023-08-06:
All miners stops and not connected to any pool and stays at 0. I try to flush again with 1.6.0 with no luck. Back to 1.5.2 and workining on ckpool. The weird thing is that from 8 miners that I have one of those still working with 1.6.0 but rest of them not.

roblatour, 2023-08-06:
I can confirm the same, with two miners, for public-pool.airdns.org (both stopped working)

thomashf, 2023-08-06:
Miners with 1.5.2 firmware using ckpool also stopped last night (just 0.0 no reboot) but luckily started again this morning.
Miner with 1.6.0 using public-pool.airdns.org stopped and rebooted every 2 minutes so I changed to ckpool after my 1.5.2 variants restarted and it also started mining.
To me it does not seem to be a firmware issue.

BitMaker-hub, 2023-08-06:
Hi, yes. Seems there is a problem with public pool currently. The pool admin is trying to fix it. I suggest you to move to ckpool

drdr34m, 2023-08-06:
The thing is that one of my miners keeps working with 1.6.0 and public pool but rest of them not

drdr34m, 2023-08-06:
> 

Ckpool with 1.6.0 not working as well and can't find the reason

roblatour, 2023-08-07:
for me public-pool.airdns.org eventually resumed working, and when I tried ckpool it was working.

closed 2023-09-01 by BitMaker-hub