BitMaker-hub/NerdMiner_v2 #133: Multiple boards

valerio-vaccaro, 2023-08-16:
Actual platfom config file shows 3 platforms:
- nerdminer_v2 - working on ttg t-display-s3
- esp32-devkit1 - not working
- ttgo t-display  - not working
(potentially wt32-sc01 and wt32-sc01 plus can be added )

Needed fixes:
- increase app space for esp32-devkit (there is no reason for a different partition table)
- remove the -U FREERTOS flag 
- remove code and dependecies not use  (e.g. display for t-display-s3 is not needed for other platform, this will need isolation between GUI component and application component like mining)

marssystems, 2023-08-16:
Take a look at this while you are waiting -

https://github.com/BitMaker-hub/ESP32_NerdMiner/tree/master

BitMaker-hub, 2023-08-16:
Thanks Valerio.
Did you take a look at dev branch. 

There is esp-wroom supported but I didn't released because there is some new functionality to add to LEDs.

Current ttgo-tdisplay-1.14 will be supported soon.

There's still some code to took off but working on it

valerio-vaccaro, 2023-08-17:
> Take a look at this while you are waiting -
> 
> https://github.com/BitMaker-hub/ESP32_NerdMiner/tree/master

Yes I know it run ... I programmed (on m5stack with firmware called han solominer) and bitmaker moved to a new board. Here the question is to use the same codebase on different platform and maybe with different graphical screens (or without a display).

sburggsx, 2023-09-09:
> @BitMaker-hub) Added support for ESP32-WROOM devices I'm having issues with esp32 wroom cause I can't find a way to program it. I have try with flasher tool and with web tool with no luck. Any one manage to program esp32 wroom?

Yes, I was just able to flash a ESP32-WROOM-32u with the following command: (run from the bin ESP32-devKit directory)

esptool.py write_flash 0xe000 0xe000_boot_app0.bin 0x8000 0x8000_partitions.bin 0x1000 0x1000_bootloader.bin 0x10000 0x10000_firmware.bin

roblatour, 2023-09-09:
@sburggsx Yes, with thanks to @BuckligerBaron 

Please see this closed issue:
https://github.com/BitMaker-hub/NerdMiner_v2/issues/144

closed 2023-09-13 by BitMaker-hub