BitMaker-hub/NerdMiner_v2 #144: online flash tool not working for the esp32 wroom-32 devkit 1

roblatour, 2023-09-01:
The flash tool at https://bitmaker-hub.github.io/diyflasher/
is not working for the esp32 wroom-32 devkit 1

you can select the board and the port, but then in about 2 seconds the webpage says the firmware was flashed but it was not - also the percent completion bars remain unchanged (all white).

BitMaker-hub, 2023-09-02:
Hi, you have to put board on boot mode. Hold boot and then ress reset or unlug and plug cable again. This should work

roblatour, 2023-09-02:
I agree, that should work - and I did try holding the boot button down and then the 'En' button but it didn't work.  Also tried unplugging it and plugging it in; still didn't work.

Here is what the screen looks like under chrome:
https://ibb.co/sRMcygb

Notice the progress bars are blank, when I flashed my T-Display S3 these turn a solid blue/purple, but with the esp32-WROOM they don't.

I can however flash the board with a simple hello world sketch from the Arduino IDE without any issues.

BuckligerBaron, 2023-09-08:
I had the same Issue on Windows. After installing this Driver [https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers](url) an additional serial Interface was installed (Com7 in my Case) and using this new Interface i could flash the Firmware. Without the Driver the ESP was recognized on Com5.
[image]

roblatour, 2023-09-08:
@BuckligerBaron Thank you - installing the driver resolved the issue for me.

closed 2023-09-08 by roblatour

Tuk2525-j, 2023-11-11:
Esp32I can't upload either.

dejota46, 2024-02-15:
> I had the same Issue on Windows. After installing this Driver [https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers](url) an additional serial Interface was installed (Com7 in my Case) and using this new Interface i could flash the Firmware. Without the Driver the ESP was recognized on Com5 by Windows. Please see Screenshot.
> 
> ![CP210](https://private-user-images.githubusercontent.com/107643900/266636848-1140e26f-9c44-4342-8923-26760330c8a0.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwMzEyMTIsIm5iZiI6MTcwODAzMDkxMiwicGF0aCI6Ii8xMDc2NDM5MDAvMjY2NjM2ODQ4LTExNDBlMjZmLTljNDQtNDM0Mi04OTIzLTI2NzYwMzMwYzhhMC5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNVQyMTAxNTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMDAwNzI5N2E4ODU5NWYzY2JmMDdmYWM0NmY5ZTBlNTk5MDVmZDJkOWRhYjBiYWJmNDM5Mzc2MGY0ZGU2MzFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.YuKNKToeLgM-We-CRfcggKLj_5BaJTX2XbgrV0XwZiM)

Hello, could you post the link again or which driver to install, because I have the same problem and I can't find the solution, thank you.