BitMaker-hub/NerdMiner_v2 #152: Add Password Field to WebUI Pool Configuration

MrLawrence, 2023-09-06:
I would like to use NerdMiner to mine altcoins with a higher chance to mine a block. A lot of pools use the password field for configuring the coin and preferred difficulty. I can't set the password in the webui, though. I would like to be able to set this as easily as the other fields.

Cj0078, 2023-09-19:
I want to know how in details to get a block on my nerd miner, and how to withdraw the btc

marssystems, 2023-09-19:
> I want to know how in details to get a block on my nerd miner, and how to withdraw the btc

When your NerdMiner finds a block the 6.25 BTC will be paid into your designated wallet automatically.

TylonHH, 2024-11-15:
but with the password field you also give specific information to some pool. So I'm looking also for the password field