BitMaker-hub/NerdMiner_v2 #173: Inability to Initially Connect to NMv2 Wifi for Configuration

Dareknotderek, 2023-09-15:
I have a `LILYGO T-Display-S3 ESP32-S3 LCD Display TTGO Development Board` that I purchased from Amazon and flashed with [nerdMiners Flash Tool](https://github.com/BitMaker-hub/NerdMiner_v2#microminers-flashtool-recommended). 

I have attempted to use my phone to scan the QR code but received `Cannot connect to network`

I've tried it on Windows 11, and received `Cannot connect to network`

I've tried it on MacOS Ventura 13.5.1 and received `Cannot connect to network`

I've copied and pasted both the SSID and Pass from the nerdminer [README.md](https://github.com/BitMaker-hub/NerdMiner_v2#readme).

Is there something else I need to do with firewalls/ports than just scan the QR code with a phone or us the SSID & Pass to initially configure the device?

Dareknotderek, 2023-09-17:
Using the ESPtool to erase all flash then reflashing has solved this.

closed 2023-09-17 by Dareknotderek

BitMaker-hub, 2023-09-18:
Fantastic, I'll add this troubleshooting to readme