BitMaker-hub/NerdMiner_v2 #179: ESP32-2432S028-R / 240x320 - nothing show at the Display - how to fix this ???

ds2k5, 2023-09-18:
Hi,
have the Model: ESP32-2432S028-R
flashed the files from: https://github.com/BitMaker-hub/NerdMiner_v2/tree/master/bin/bin%20ESP32-2432S028R

```
esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 0x1000_bootloader.bin 
esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x8000 0x8000_partitions.bin 
esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x10000 0x10000_firmware.bin 
esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0xe000 0xe000_boot_app0.bin 

```
can see the WiFI:  NerdMinerAP

How can I fix the Display issue ?

Thanks

ds2k5, 2023-09-18:
did online flash with: https://bitmaker-hub.github.io/diyflasher/
the display works

closed 2023-09-18 by ds2k5

izhzz, 2023-11-05:
Guys, I have the same units. When I connect it to flash it, it just reboots. If I find it and connect to it, it gets a timeout. I have multiple devices and all are the same. What am I doing wrong? It just reboots into the Menu with tabs Controls > Visuals > Selectors. I flashed a S3 before, which was very easy compared to this.