BitMaker-hub/NerdMiner_v2 #180: Brand new LILYGO T-Display-S3 can't mine. 0.00KH/s

horrorfodrasz, 2023-09-18:
Hi,
I bought 2pcs LILYGO T-Display S3. I flashed it with online flasher (https://bitmaker-hub.github.io/diyflasher/). 
Device is working. I was able to connect it to Wifi, I can switch screens, etc, but the hashrate is 0.00KH/s.
I tried it with 1.6.1, 1.6.0 and 1.5.2 fw, too. The result was the same.
My both T-Display-S3 can't mine.  Hashrate is 0.00 in case of both boards.
- I tried another powersupply
- I tried more pools
- I tried another type BTC address
- I tried to connect to another wifi
- I tried to flash with online ESP tool ("bin LILYGO TDisplay S3" folder), 
the result is the same.
What can cause this issue?
Thank you for the help,

Dareknotderek, 2023-09-18:
Have you tried to erase all flash then reflash it?

The esp tool has an erase all option to completely clean the s3 before flashing.

horrorfodrasz, 2023-09-18:
I tried. I have erased the whole flash and reprogrammed with ESP tool. It seems it is working but the hashrate is 0.00...

Dareknotderek, 2023-09-19:
Which pool? As long as it's a low difficulty pool you should be fine. Some pools might not send it shares if it's that low difficulty. 

Either way, I'm fairly new to this still so hopefully other nerdminers can come help out!

horrorfodrasz, 2023-09-19:
I have tried all pools that can be found on Nerdminer's Github info page....

horrorfodrasz, 2023-09-20:
I think I am close to the source of the issue. I connected the Miner to my phone's hotspot and it is working now. I think it can not mine through my Asus router. When I connect it to my main home router I think it can reach the internet because it can see BTC stats, but mining doesn't start. 
What settings should I check in router? 
Maybe it is not compatible with 802.11ax and/or WPA2-Personal?

BitMaker-hub, 2023-09-20:
Hi maybe is blocking some TCP connections. Stratum protocol is using TCP/IP protocol and defined port (by pool) 
On the standard default pool, port is 21496. Check if your router is Blocking those ports

stevehansen, 2023-09-20:
same issue here, also using an Asus router, wasn't able to test another connection yet, can't find anything specific to it blocking that port

stevehansen, 2023-09-20:
disable the TwoWay IPS in the AI protection menu fixes the issue
[image]

found it via https://www.reddit.com/r/MoneroMining/comments/m1g0y1/cannot_reach_any_pool_without_tls/

it appears to be caused by the pool not supporting TLS

horrorfodrasz, 2023-09-20:
Thanks! That was the solution.