BitMaker-hub/NerdMiner_v2 #188: [Question] - Is it possible to combine multiple ESP32 DevKits to work together?

PsySc0rpi0n, 2023-09-25:
Just wondering if it is possible to use multiple ESP32 working together combining their hash power as if it was a single machine doing all the work.

For instance, a single ESP32 is doing arounf 50kH/s. If I have 5 of them, would it be possible to be working at around 250kH/s as a single device?

dsbaha, 2023-10-16:
I've been using a stratum proxy to accomplish this.  Although the stratum proxy I'm using has bugs.  I'm thinking of just quickly writing a go-lang based stratum proxy.

TylonHH, 2023-10-17:
What about an 50KH Pool for all slow miner?

williammagno, 2024-04-07:
Hi, someone solved this?

sany3001, 2024-04-07:
Probably different solution, but I am using ESP32-2432S028R to see total hashrate of all my miners. I am using same BTC address on all of them. But still, each of them is mining solo.

williammagno, 2024-04-07:
> Probably different solution, but I am using ESP32-2432S028R to see total hashrate of all my miners. I am using same BTC address on all of them. But still, each of them is mining solo.

Can you share how you made to show the total hashrate on ESP32-2432S028R? Thanks :)

paelzer, 2024-04-07:
ESP32-2432S028R devices have 320x240px resolution screens. These ones show total hash rate by default in lower right corner on the main screen.