BitMaker-hub/NerdMiner_v2 #191: [Question]: Is it possible with ESP32-C3

SergioMuinhos, 2023-09-26:
Hi, 
I have just buy the [LILYGO® T-Zigbee T-ZigbeePA ESP32-C3 TLSR8258 Zigbee](https://es.aliexpress.com/item/1005004182389804.html?spm=a2g0o.order_list.order_list_main.10.1cc8194d11Xhaz&gatewayAdapt=glo2esp)
And i just ask if can i flash this proyect in this device.
Regards.

nitroxgas, 2023-10-05:
I tried, got max 20KH/s;
C3 was build with power saving in mind. Low processing specs compared to other ESP32.

bgramaje, 2024-03-11:
Hello everyone. I have tried on deploying it on a ESP32-C3 but the serial monitor returns the following: 
`Checksum failure. Calculated 0x2c stored 0x46
20:53:59.567 -> ets_main.c 333 
20:54:00.863 -> ESP-ROM:esp32c3-api1-20210207
20:54:00.863 -> Build:Feb  7 2021
20:54:00.863 -> rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
20:54:00.863 -> Saved PC:0x40047ed2
20:54:00.863 -> SPIWP:0xee
20:54:00.863 -> mode:DIO, clock div:1
20:54:00.863 -> load:0x3fcd5810,len:0x438
20:54:00.863 -> load:0x403cc710,len:0x918
`

Any guess? Thanks