BitMaker-hub/NerdMiner_v2 #192: KH/s of every board?

ccp42, 2023-09-27:
Is it possible to know how many KHs give every board? Are they similar? In youtube videos show ESP-32 gives 55 KHs and LILYGO T-Display S3  22 KHs...
? LILYGO T-Display S3 
? ESP32-WROOM-32, ESP32-Devkit1.. 
? LILYGO T-QT pro 
? LILYGO T-Display 1.14 
? LILYGO T-Display S3 AMOLED 
? LILYGO T-Dongle S3 
? ESP32-2432S028R 2,8"
Thank you.

BitMaker-hub, 2023-09-28:
Hi, thanks thats a cool idea, I'll publish at readme. 
With latest sha lib i got more hashrate

closed 2023-09-28 by BitMaker-hub

matteocrippa, 2023-10-28:
@BitMaker-hub Did you had a chance to look into that? I was curious to know how different is the performance of a ESP32-wroom vs S3
I do expect S3 should be somehow better performing but checking at the stats does not seems so

BitMaker-hub, 2023-10-28:
Using the standard NerdSha lib:

56Khs - LILYGO T-Display S3
45Khs -  ESP32-WROOM-32, ESP32-Devkit1..
56Khs -  LILYGO T-QT pro
45Khs -  LILYGO T-Display 1.14
56Khs -  LILYGO T-Display S3 AMOLED
56Khs -  LILYGO T-Dongle S3
45Khs -  ESP32-2432S028R 2,8"

Using the new unreleased sha lib (this November will release):
78Khs - LILYGO T-Display S3
55Khs -  ESP32-WROOM-32, ESP32-Devkit1..
78Khs -  LILYGO T-QT pro
55Khs -  LILYGO T-Display 1.14
78Khs -  LILYGO T-Display S3 AMOLED
78Khs -  LILYGO T-Dongle S3
55Khs -  ESP32-2432S028R 2,8"

rik43, 2025-07-16:
what about ESP32-C3 ?