BitMaker-hub/NerdMiner_v2 #198: Sreen issue Powering with 5vDC

lilsow, 2023-10-06:
Hi ! I have 3 Lilygo T display S3 NerdMiners, One brought flashed with NerdMiner on Bitwater's website and 2 brought on Lilygo's website.
I try to design a PCB to power my NerdMiners with a 5v DC power supply. When powered like this, the one brought on bitwater (flashed since with the new firmware) works like a charm, the two others seems to work (can see them working on Publicpool) but nothing displays on the screen. When i plug them on a USB alimentation, all devices has screen on. I Installed Lilygo T Display S3 original firmware to try, when powering on 5vDC All screens are working !
So asking myself some questions, flashed the 3 ESP32 with the same firmware so why do they have different behaviour ?
Saw some informations related to PIN 15 than needs to be set to HIGH, but not really sure of the procedure, tried to add the code to the ESP then flashing back to nerdminer but doesn't seem to change anything.
Saw few shemes of PCB's on ghithub, and seems totally possible to power Nerdminers with 5vDC over pins. Does someone has an idea why one could be working and the others dont ?
Thanks for your help !

Lil Slow

sany3001, 2023-10-07:
Hi Lil!
I guess I am runnig same HW (have got in total 8 pieces). On one piece I am observing probably same situation as you are describing. Both red and green LEDs on the bottom are always on but it case I am supplying my boards via pin headers display of one of them is off. When supplied via USB-C it goes on and works as the other.

 I need you help how to check from pool perspective it is mining. Can you advice? I am using default pool (Public-pool.io).

I am using FW from this site: https://github.com/BitMaker-hub/NerdMiner_v2 (v1.6.2).

Regarding PIN 15 on Lilygo. Have you tried to tie 5V to it (to get it into high state)? Where have you found information that it needs to be HIGH?

5V power supply definitely works, am using this rackmount from Sulfuroid: https://twitter.com/sulfuroid/status/1673794180995268613.

lilsow, 2023-10-07:
Hi Sany! Thank you for your answer!
To see if my NerdMiners are working I simply check on https://web.public-pool.io/#/ researching for my BTC address and seeing how many devices mining on this address. Nice trick to check your devices working from anywhere !
Using same FW as you on my devices.
Saw info on pin 15 on few forums and on this video https://www.youtube.com/watch?v=TNFJQKrsoOQ&ab_channel=PeterSchneider
don't know if legit or not.
Inspired my project from this board ! So a bit confused to face this problem. All your devices were brought empty ? Or flashed with NerdMiner Firmware ?

sany3001, 2023-10-08:
Hi again! Thanks for hint with monitoring and link to video.
Yes, all my devices were bought empty and I have flashed them with same FW. Initial was not 1.6.2 but currently all are on the latest release.

lilsow, 2023-10-08:
Have you installed other things on your Lilygos before Nerdminer Soft ? Does the problem was here even with old version ?

sany3001, 2023-10-09:
I have not installed any other SW on Lilugos. First SW on it was NerdMiner.
I am pretty sure that this problem was there with previous version (probably 1.6.1) as I was talking about this issue with my fried and when 1.6.2 was out he told me that maybe this problem will be fixed.

What I want to try now is to monitor is miner is connected to Wi-Fi at all in this situation. But all my miner are off now as I am waiting for heat sinks. I would like also to see ESP temperature on NerdMiner screen to check if heat sink installation does desired effect or if fan installation is still needed. There is open PR for that, but already for 3 months and doesn't seem to progress. So I will try to flash this change on my HW and test it at least.

sany3001, 2023-10-09:
Have you looked into code? What is pin GPIO10 (pin 15) used for? According to https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/T_Display_S3.pdf this pin is not used, so setting it to HIGH will have no effect.