BitMaker-hub/NerdMiner_v2 #199: ESP32-S3-WRoom-1 programming not working

Keksi71, 2023-10-11:
Dear all, i bought some of this boards for programming with the miner.bins for the S3 Wroom.I tried different ways but nothing works,
I bought this one.
[https://de.aliexpress.com/item/1005005767192637.html](url)
Anyone knows what i have to do?

BR Keksi

Tuk2525-j, 2023-11-11:
Teach me, I don't understand.

Keksi71, 2023-11-11:
Hello, i tried different ways to get this two esp 32 running with the firmware and bootfiles for the nerdminer.

Tasmota and esphome flashing works. Do you have an idea what i can try?

 

BR Oli

PSLLSP, 2024-01-21:
I have several ESP-WROOM-32 modules and some of them cannot be flashed WebFlasher, I confirm that. I do not know why, maybe that some signal from UART (DTR or RTS) is not connected to ESP32 or these signals are connected in a "wrong" way.

Anyway, I was able to flash those with [esptool.py](https://docs.espressif.com/projects/esptool/en/latest/esp32/) that is command line tool and it needs several parameters, 4 pieces of firmware has to be flashed... Firmware is in NerdMiner git repository.

sany3001, 2024-01-21:
I have opened similar topic: https://github.com/BitMaker-hub/NerdMiner_v2/issues/298.
My device on development board is ESP32-WROOM-1U. The board looks very similar to yours. You will succeed, try it!