BitMaker-hub/NerdMiner_v2 #202: [Feature] ESP32-S2FN4R2 (aka Lolin S2 mini) support

julio-matarranz, 2023-10-14:
Dear,

Could someone handle ESP32-S2FN4R2 please ?

It's a cheap and widespread ESP32-S2 module. It would be nice to have NerdMiner V2 on it.
Thank you in advance.

elmo128, 2023-10-19:
Hi Julio,
it is possible to build Nerdminer for boards, using the IC you specified (at least with basic settings).
Unfortunateley, i don't own such a board and can't test it.

elmo128, 2023-10-27:
Hi @julio-matarranz,
thanks for testing.
Unfortunateley I can't reproduce the issues on the platform available to me.
I applied some minor changes to the config file. Additionally i recommend you to erase the flash memory of your esp32 manually before uploading the firmware, to get rid of config data already stored to the device. Please also make sure, that no other Wifi access point named "NerdMinerAP" is in range. If this does not help, please try the new branch https://github.com/elmo128/NerdMiner_v2/tree/lolinS2-Mini-PoC-Set2nv. It does not make use of SPIFFS anymore, that might be incomaptible with your board. It is only a quick and dirty implementation as proof of concept.

julio-matarranz, 2023-10-30:
Dear,
Thank you for your help but I think my S2 mini can go to trash. After many tries the wifi seem to be broken on that cheap clone.
Thank you again and sorry for your time

elmo128, 2023-10-31:
Hello@ julio-matarranz,
You're welcome.
Thanks for your contribution.
There are a lot of applications for esp32 based devices, some of which don't even require wifi....

julio-matarranz, 2023-10-31:
@elmo128 I was kidding about throwing it away, I will try to fix it one day if there is a good fix but for now it sound pretty experimental with adding a random capacitor.
I bought some ESP32-S3 from Lilygo, they aren't cheap but I know why Lolin S2 mini clones are cheap... 🤣

Your first code for ESP32S2 should be enough if my hardware weren't faulty. 

I don't know what capacitor I should buy on AliExpress to fix the WiFi but I would do that soon ish and let you know. ☺️

Thank you again

elmo128, 2023-11-01:
Hi @julio-matarranz,
:joy:
> I don't know what capacitor I should buy on AliExpress

> [tzapu/WiFiManager#1478](https://github.com/tzapu/WiFiManager/issues/1478)

They link to the espressif forum, where possible hardware issues and fixes are discussed: https://esp32.com/viewtopic.php?f=19&t=28506 Other sources can be the documentation (datasheets, manuals, appnotes, etc.) provided by the manufacturer for the IC or eval boards.
I probably don't have to write this, but i recommend you to try some other programs of your choice (especially wifi demos/examples) to confirm, that your board is indeed faulty, before messing around.

JohnTaipei, 2023-11-12:
@elmo128 
Hello Sir:
Could you create .BIN files for S2 mini board??
Thanks

elmo128, 2023-11-13:
@JohnTaipei,
hello Sir, thanks for your interest. Although i have never uploaded binaries before and am not 100% sure about which files are required, i added some binaries to each branch starting with lolinS2-Mini.
Please read the discussion above, to find out more about issues. I can't give you any guarantees, not even that i uploaded all files that are required or that the addresses contained in most file names are right.
Thus i recommend you to install vcstudio and platformio, if you run into errors, to make sure that everything has been uploaded in the right way.
Have a lot of success!

JohnTaipei, 2023-11-17:
@elmo128 
Thank you so much Sir :D 
I will try it and hope it works

ben-barnard, 2024-01-06:
Any update on this? Would love to use mine.

closed 2024-04-09 by BitMaker-hub

gungriffonbr, 2024-12-21:
i dont find any bin files for Wemos Lolin S2 Mini

PR #334 (2024-04-09): feat : adding support to lilygo s3 amoled touch 78khs & waveshare ESP32-S3-GEEK dongle 78khs  & weact ESP32-D0WD-V3 57khs  & wemos s2 mini 38khs fixes #202 and disable/enable led on no display board
adding support to lilygo s3 amoled touch 78khs
adding support to waveshare ESP32-S3-GEEK dongle 78khs
adding support to weact ESP32-D0WD-V3 57khs 
adding wemos s2 mini 38khs fixes #202
disable/enable led on no display board
update supported board documentation
adding firmwares
commit 00d89a5e (2024-04-09): feat : adding support to wemos s2 mini 38khs fixes #202
commit 1a37be7a (2024-04-09): Merge pull request #334 from julio-matarranz/feat/esp32-d0wd-and-s2-mini

feat : adding support to lilygo s3 amoled touch 78khs & waveshare ESP32-S3-GEEK dongle 78khs  & weact ESP32-D0WD-V3 57khs  & wemos s2 mini 38khs fixes #202 and disable/enable led on no display board