BitMaker-hub/NerdMiner_v2 #214: can not display the btc price

arcsin3x, 2023-11-04:
[image]

arcsin3x, 2023-11-04:
Can you increase the font size of the price？

Goohst, 2023-11-04:
I can't connect to this API. Can you add the BTC price API to the configuration file and fill it in myself? Can you also write the refresh time of the BTC price into the configuration file so that I can set it myself?

StarMikeMichael, 2023-11-26:
Yesterday the price resumed showing, and is still showing accurate.  Also, for the first time the the field 32Bits Shares: shows a quantity 3.  

Can any body please tell me what the 32Bits Shares is counting?  Still waiting for my valid blocks to start counting.  But in the mean time, I had ZERO 32 Bit shares for about a week, now that the BTC $0 Price display fixed itself, I show my first THREE 32Bit Shares, and an uptime of two and a half days.  

 (maybe the pool softtwhere was updated, I havn't checked the pool web site yet.)
 I'm still trying to find where in the nerdminer code he gets the price from......  
 
 Totally a fun project!

arcsin3x, 2023-11-26:
> Yesterday the price resumed showing, and is still showing accurate. Also, for the first time the the field 32Bits Shares: shows a quantity 3.
>
> Can any body please tell me what the 32Bits Shares is [...]
Hi,where r u from? Some country can not connect api.

BitMaker-hub, 2023-11-29:
Guys, thanks for your contributions, your changes were merged.
So now everybody will see the price correctly even in China

Regardint 32 bit share means, how many shares were generated by this microminer that has at least 32 bits of zeros on its hash

closed 2023-11-29 by BitMaker-hub

arcsin3x, 2023-11-30:
我再提供一种可以连接API，好像无次数限制，     https://min-api.cryptocompare.com/data/price?fsym=BTC&tsyms=USD  直接修改API地址为以上地址，然后修改代码 if (doc.containsKey("USD")) bitcoin_price = doc["USD"];      （在src/monitor.h和src/monitor.cpp两个文件中）或者直接使用我的fork

arcsin3x, 2023-11-30:
> Wait, the one you gave me has limit restrictions? I don't want to be changing it a lot. Changing price every minute is like 1440 request day is this too much????

I'm not sure if there are any restrictions, but the last API address price update took quite some time.This one has a fast update frequency.