BitMaker-hub/NerdMiner_v2 #215: I can not connect NerdMinerAP.

BCProgrammingTH, 2023-11-04:
I also installed the firmware. bitmaker-hub.github.io I found NerdMinerAP but MineYourCoin password is incorrect.

PSLLSP, 2023-11-07:
I flashed firmware to [T-Dongle-S3](https://www.lilygo.cc/products/t-dongle-s3) with "easy to use" tool at  [NM DIY Flasher](https://nerdminer.satoshistore.io/). Instruction on that page could be better, I miss instruction how to enter "boot mode". Other issue was with Chromium web browser running on Linux, it cannot access "serial port".  I installed "Google Chrome" and that one can see ports and just works...

OK, so my T-Dongle-S3 was flashed and I see message `Use WiFi for setup...` on OLED display. I see there is a new AP with name `NerdMinerAP` but I cannot find any information about password, it is a secret, not documented! :-( I Tried **MineYourCoin** and several others but it doesn't work and I cannot connect...

rangelrmorais, 2024-01-19:
Not working, the password is incorrect even with QRCode. Impossible to connect to NerdMinerAP with MineYourCoins.

My board: LILYGO® T-Display-S3

PSLLSP, 2024-01-19:
> Not working, the password is incorrect even with QRCode. Impossible to connect to NerdMinerAP with MineYourCoins.

Try to reflash firmware in your device...

[WebFlasher - bitronics](https://flasher.bitronics.store/) or [WebFlasher - satoshistore](https://nerdminer.satoshistore.io/)

rangelrmorais, 2024-04-19:
> > Not working, the password is incorrect even with QRCode. Impossible to connect to NerdMinerAP with MineYourCoins.
> 
> Try to reflash firmware in your device...
> 
> [WebFlasher - bitronics](https://flasher.bitronics.store/) or [WebFlasher - satoshistore](https://nerdminer.satoshistore.io/)
> 
> Your device has display and it shows QR code. What is password in that QR code? Can you decode it?? Anyway, when you flash your device with Webflasher, it should have standard SSID and password...
> 
> There is similar device (a clone) on the market that has different SSID and WiFi password. [JingleMiner](https://www.jinglemining.com/blogs/after-sale-faq-tutorial/jingleminer-configuration-flashing-tutorial) has SSID `JingleMiner` and WiFi password is `jinglemining` (displayed on the screen) or `JingleMinerBTC` (stored in QR code that is displayed on the screen). Something is wrong there, password stored in the QR code doesn't match password on the screen...
> 
> Good barcode scanner/decoder for Android is [Binary Eye](https://play.google.com/store/apps/details?id=de.markusfisch.android.binaryeye), it can help with QR code troubleshooting...

Flashed and reflashed over and over again and it wouldnt connect at all. The password displaying on the screen is the default MineYourCoins.

Oberlandix, 2024-04-29:
Hello everyone,

I am "glad" to have found this issue here as I have also been struggling with exactly the same problem for several hours. I have ordered, flashed and set up three Nerdminers. With two of them I had no problem at all - with the third miner I've been stuck for hours now because the password is supposedly wrong. I have already reflashed several times. With the ESP tool and also via the "Chrome browser". The Nerdminer shows the access point with the SSID NerdMinerAP - when logging in via mobile phone, tablet, PC etc., I always get the message that the password (MineYourCoins) is incorrect. However, the Storage.sh also contains exactly this password.

I'm losing hours here and am about to freak out because it worked so easily with the first two miners. If anyone else has any ideas, please let me know.

Best regards and a little remaining hope, Oberlandix

PSLLSP, 2024-04-30:
There is a way to backup firmware to a file and check that file for a password. Firmware could be saved to file with `esptool`, this command saves 8 MB of flash:

```
esptool.py read_flash 0 0x800000 NM-firmware-backup.bin
```

Firmware could be restored; you do not need this but it is good to know this option... 

```
esptool.py write_flash 0 NM-firmware-backup.bin
```

Then you can try to search for known passwords and you can compare NerdMiner that is working with  the one that rejects password... On Linux command `strings` will print all printable strings in a file, then you can use `grep` to search for password.

Other hint is to erase the device and flash it again. This is the command to erase whole flash:

```
esptool erase_flash
```

Mrtechit10121, 2024-06-28:
I have this issue right now, been trying for hours to get in. 
I have the dongle s3, plug it in, i go to connect to NerdMinerAP with password MineYourCoins and it says wrong password. I have flashed using the online flash tool many times, same thing....

Any idea how to fix? I am new to this sorta thing but willing to learn.

Mrtechit10121, 2024-06-28:
Wow. As i typed this, I flashed yet again (22 times i flashed btw) I flashed using the github 1.7 firmware using Bittronics DIY flasher again for the dongle. The AP password worked this time and I was able to set it up. 
I did plug it in to the other computer, with arduino ide on it and pressed a bunch of buttons because I have never programmed or used something like that before.... i hit the reset button a lot lol. Either way, it works now. solid 75Kh